ARMmbed/mbed-client-example-6lowpan #23: Example fails to compile for F429.

markus-becker-tridonic-com, 2015-09-15:
The compilation for the F429 fails with 

```
In file included from mbed-client-example-
6lowpan/yotta_modules/sal/source/test/socket_abstract_test.cpp:19:0:
mbed-client-example-6lowpan/yotta_modules/sal/mbed-net-socket-abstract/socket_api.h:95:19: fatal error: cmsis.h: No such file or directory
 #include "cmsis.h"
```

For the K64F this works, since that target has a dependency on lwip, which adds the correct include paths. Shall I prepare a pull request to add that dependency for F429 as well, or should that lwip dependency be removed from the 6lowpan example? Then this target needs to add (at least) the includes for ```mbed-client-example-6lowpan/yotta_modules/cmsis-core-stm32f4/cmsis-core-stm32f4```.

artokin, 2015-09-15:
At the moment this example application supports only K64F and unfortunately the lwip stack is included to this application even though it is not really needed by this example application. @bremoran would you please help with the this issue?

markus-becker-tridonic-com, 2015-09-15:
Forgot to say that we have this solved by including lwip for F429 and could contribute that. The correct solution would however be to have the includes corrected and not depend on lwip.

bremoran, 2015-09-15:
@artokin Doing this correctly requires some changes to the k64f target and the socket abstraction layer.  I'll look into it.

hwaldman, 2015-09-18:
I get the following errors when trying to build for K64f with gcc:
C:\work\temp\MBED\CodeDrops\beta\thread_example\mbed-client-example-6lowpan>yt b
uild > scrap.txt
error: mbed-drivers does not meet specification ~0.7.0 required by mbed-6lowpan-
eventloop-adaptor
error: minar does not meet specification ~0.7.0 required by sockets
error: core-util does not meet specification ~0.1.2 required by sockets

and

FAILED: C:\PROGRA~2\GNUTOO~1\4BD65~1.920\bin\AR10B2~1.EXE  -DYOTTA_COMPONENT_VERSION=\"0.3.5\" -fno-exceptions -fno-unwind-tables -ffunction-sections -fdata-sections -Wall -Wextra -fno-rtti -fno-threadsafe-statics -mcpu=cortex-m4 -mthumb -D__thumb2__ -Os -g -DNDEBUG -IC:/work/temp/MBED/CodeDrops/beta/thread_example/mbed-client-example-6lowpan -IC:/work/temp/MBED/CodeDrops/beta/thread_example/mbed-client-example-6lowpan/yotta_modules/mbed-client -IC:/work/temp/MBED/CodeDrops/beta/thread_example/mbed-client-example-6lowpan/yotta_modules/atmel-rf-driver -IC:/work/temp/MBED/CodeDrops/beta/thread_example/mbed-client-example-6lowpan/yotta_modules/mbed-mesh-api -IC:/work/temp/MBED/CodeDrops/beta/thread_example/mbed-client-example-6lowpan/yotta_modules/mbed-client-c -IC:/work/temp/MBED/CodeDrops/beta/thread_example/mbed-client-example-6lowpan/yotta_modules/mbed-client-mbed-os -IC:/work/temp/MBED/CodeDrops/beta/thread_example/mbed-client-example-6lowpan/yotta_modules/mbed-client-mbedtls -IC:/work/temp/MBED/CodeDrops/beta/thread_example/mbed-client-example-6lowpan/yotta_modules/nanostack-libservice -IC:/work/temp/MBED/CodeDrops/beta/thread_example/mbed-client-example-6lowpan/yotta_modules/mbed-drivers -IC:/work/temp/MBED/CodeDrops/beta/thread_example/mbed-client-example-6lowpan/yotta_modules/sockets -IC:/work/temp/MBED/CodeDrops/beta/thread_example/mbed-client-example-6lowpan/yotta_modules/mbed-hal -IC:/work/temp/MBED/CodeDrops/beta/thread_example/mbed-client-example-6lowpan/yotta_modules/cmsis-core -IC:/work/temp/MBED/CodeDrops/beta/thread_example/mbed-client-example-6lowpan/yotta_modules/ualloc -IC:/work/temp/MBED/CodeDrops/beta/thread_example/mbed-client-example-6lowpan/yotta_modules/minar -IC:/work/temp/MBED/CodeDrops/beta/thread_example/mbed-client-example-6lowpan/yotta_modules/core-util -IC:/work/temp/MBED/CodeDrops/beta/thread_example/mbed-client-example-6lowpan/yotta_modules/mbed-hal-freescale -IC:/work/temp/MBED/CodeDrops/beta/thread_example/mbed-client-example-6lowpan/yotta_modules/mbed-hal-ksdk-mcu -IC:/work/temp/MBED/CodeDrops/beta/thread_example/mbed-client-example-6lowpan/yotta_modules/uvisor-lib -IC:/work/temp/MBED/CodeDrops/beta/thread_example/mbed-client-example-6lowpan/yotta_modules/mbed-hal-k64f -IC:/work/temp/MBED/CodeDrops/beta/thread_example/mbed-client-example-6lowpan/yotta_modules/mbed-hal-frdm-k64f -IC:/work/temp/MBED/CodeDrops/beta/thread_example/mbed-client-example-6lowpan/yotta_modules/cmsis-core-freescale -IC:/work/temp/MBED/CodeDrops/beta/thread_example/mbed-client-example-6lowpan/yotta_modules/cmsis-core-k64f -IC:/work/temp/MBED/CodeDrops/beta/thread_example/mbed-client-example-6lowpan/yotta_modules/dlmalloc -IC:/work/temp/MBED/CodeDrops/beta/thread_example/mbed-client-example-6lowpan/yotta_modules/compiler-polyfill -IC:/work/temp/MBED/CodeDrops/beta/thread_example/mbed-client-example-6lowpan/yotta_modules/minar-platform -IC:/work/temp/MBED/CodeDrops/beta/thread_example/mbed-client-example-6lowpan/yotta_modules/minar-platform-mbed -IC:/work/temp/MBED/CodeDrops/beta/thread_example/mbed-client-example-6lowpan/yotta_modules/sal -IC:/work/temp/MBED/CodeDrops/beta/thread_example/mbed-client-example-6lowpan/yotta_modules/sal-stack-lwip -IC:/work/temp/MBED/CodeDrops/beta/thread_example/mbed-client-example-6lowpan/yotta_modules/sal-driver-lwip-k64f-eth -IC:/work/temp/MBED/CodeDrops/beta/thread_example/mbed-client-example-6lowpan/yotta_modules/sal-iface-eth -IC:/work/temp/MBED/CodeDrops/beta/thread_example/mbed-client-example-6lowpan/yotta_modules/mbedtls -IC:/work/temp/MBED/CodeDrops/beta/thread_example/mbed-client-example-6lowpan/yotta_modules/sal-stack-nanostack -IC:/work/temp/MBED/CodeDrops/beta/thread_example/mbed-client-example-6lowpan/yotta_modules/sal-stack-nanostack-eventloop -IC:/work/temp/MBED/CodeDrops/beta/thread_example/mbed-client-example-6lowpan/yotta_modules/nanostack-randlib -IC:/work/temp/MBED/CodeDrops/beta/thread_example/mbed-client-example-6lowpan/yotta_modules/mbed-6lowpan-eventloop-adaptor -IC:/work/temp/MBED/CodeDrops/beta/thread_example/mbed-client-example-6lowpan/yotta_modules/sal-iface-6lowpan -IC:/work/temp/MBED/CodeDrops/beta/thread_example/mbed-client-example-6lowpan/yotta_modules/sal-stack-lwip/mbed-net-lwip -IC:/work/temp/MBED/CodeDrops/beta/thread_example/mbed-client-example-6lowpan/yotta_modules/sal-stack-lwip/mbed-net-lwip/arch -IC:/work/temp/MBED/CodeDrops/beta/thread_example/mbed-client-example-6lowpan/yotta_modules/sal-stack-lwip/mbed-net-lwip/lwip -IC:/work/temp/MBED/CodeDrops/beta/thread_example/mbed-client-example-6lowpan/yotta_modules/sal-stack-lwip/mbed-net-lwip/lwip/include -IC:/work/temp/MBED/CodeDrops/beta/thread_example/mbed-client-example-6lowpan/yotta_modules/sal-stack-lwip/mbed-net-lwip/lwip/include/ipv4 -IC:/work/temp/MBED/CodeDrops/beta/thread_example/mbed-client-example-6lowpan/yotta_modules/sal-stack-lwip/mbed-net-lwip/lwip/include/netif -IC:/work/temp/MBED/CodeDrops/beta/thread_example/mbed-client-example-6lowpan/yotta_modules/sal-driver-lwip-k64f-eth/mbed-net-lwip-k64f -IC:/work/temp/MBED/CodeDrops/beta/thread_example/mbed-clie [... cut ...] ockets/source/v0/TCPListener.cpp
In file included from C:/work/temp/MBED/CodeDrops/beta/thread_example/mbed-client-example-6lowpan/yotta_modules/sockets/source/v0/TCPListener.cpp:18:0:

C:/work/temp/MBED/CodeDrops/beta/thread_example/mbed-client-example-6lowpan/yotta_modules/sockets/mbed-net-sockets/v0/TCPListener.h:22:39: fatal error: mbed-util/FunctionPointer.h: No such file or directory

 #include "mbed-util/FunctionPointer.h"

                                       ^

compilation terminated.

bremoran, 2015-09-21:
@hwaldman, I will need to see the output of ```yt ls``` to be able to help any further.

hwaldman, 2015-09-21:
I re-posted this as a new issue and it got fixed. Thanks.

artokin, 2015-09-22:
Here is link to the re-posted issue https://github.com/ARMmbed/mbed-client-example-6lowpan/issues/24.

rainierwolfcastle, 2015-10-15:
http://jira-dev0.cambridge.arm.com/browse/ONME-1698

rainierwolfcastle, 2015-10-16:
ARM Internal Ref: ONME-1990

bremoran, 2015-10-22:
I have now changed SAL to import dependencies via yotta config.  LwIP can be disabled by setting an appropriate flag in application level configuration:

```JSON
"mbed-os" : {
  "net" : {
    "stacks" : {
      "lwip" : false,
      "nanostack" : true
    }
  }
}
```

The cmsis dependency is still a problem.  There is an open PR for this issue: https://github.com/ARMmbed/sal/pull/31

mlnx, 2015-12-02:
@bremoran could you please look at this ticket and provide an update?

bremoran, 2015-12-02:
The PR was merged on Oct 27. This issue should be resolved now.

artokin, 2015-12-04:
@markus-becker-tridonic-com, can this issue be closed?

markus-becker-tridonic-com, 2015-12-10:
@bremoran @bcostm @autopulated The cmsis part of the problem '6LoWPAN on F4' is solved. Other changes are still required to really compile it, see https://github.com/ARMmbed/target-st-stm32f429i-disco/issues/16#issuecomment-161041151.

closed 2015-12-10 by markus-becker-tridonic-com