ARMmbed/mbed-client-example-6lowpan #24: fail to build for K64F

hwaldman, 2015-09-18:
I get the following errors when trying to build for K64f with gcc:
 C:\work\temp\MBED\CodeDrops\beta\thread_example\mbed-client-example-6lowpan>yt b
 uild > scrap.txt
 error: mbed-drivers does not meet specification ~0.7.0 required by mbed-6lowpan-
 eventloop-adaptor
 error: minar does not meet specification ~0.7.0 required by sockets
 error: core-util does not meet specification ~0.1.2 required by sockets

and

FAILED: C:\PROGRA~2\GNUTOO~1\4BD65~1.920\bin\AR10B2~1.EXE -DYOTTA_COMPONENT_VERSION=\"0.3.5\" -fno-exceptions -fno-unwind-tables -ffunction-sections -fdata-sections -Wall -Wextra -fno-rtti -fno-threadsafe-statics -mcpu=cortex-m4 -mthumb -D__thumb2__ -Os -g -DNDEBUG -IC:/work/temp/MBED/CodeDrops/beta/thread_example/mbed-client-example-6lowpan -IC:/work/temp/MBED/CodeDrops/beta/thread_example/mbed-client-example-6lowpan/yotta_modules/mbed-client -IC:/work/temp/MBED/CodeDrops/beta/thread_example/mbed-client-example-6lowpan/yotta_modules/atmel-rf-driver -IC:/work/temp/MBED/CodeDrops/beta/thread_example/mbed-client-example-6lowpan/yotta_modules/mbed-mesh-api -IC:/work/temp/MBED/CodeDrops/beta/thread_example/mbed-client-example-6lowpan/yotta_modules/mbed-client-c -IC:/work/temp/MBED/CodeDrops/beta/thread_example/mbed-client-example-6lowpan/yotta_modules/mbed-client-mbed-os -IC:/work/temp/MBED/CodeDrops/beta/thread_example/mbed-client-example-6lowpan/yotta_modules/mbed-client-mbedtls -IC:/work/temp/MBED/CodeDrops/beta/thread_example/mbed-client-example-6lowpan/yotta_modules/nanostack-libservice -IC:/work/temp/MBED/CodeDrops/beta/thread_example/mbed-client-example-6lowpan/yotta_modules/mbed-drivers -IC:/work/temp/MBED/CodeDrops/beta/thread_example/mbed-client-example-6lowpan/yotta_modules/sockets -IC:/work/temp/MBED/CodeDrops/beta/thread_example/mbed-client-example-6lowpan/yotta_modules/mbed-hal -IC:/work/temp/MBED/CodeDrops/beta/thread_example/mbed-client-example-6lowpan/yotta_modules/cmsis-core -IC:/work/temp/MBED/CodeDrops/beta/thread_example/mbed-client-example-6lowpan/yotta_modules/ualloc -IC:/work/temp/MBED/CodeDrops/beta/thread_example/mbed-client-example-6lowpan/yotta_modules/minar -IC:/work/temp/MBED/CodeDrops/beta/thread_example/mbed-client-example-6lowpan/yotta_modules/core-util -IC:/work/temp/MBED/CodeDrops/beta/thread_example/mbed-client-example-6lowpan/yotta_modules/mbed-hal-freescale -IC:/work/temp/MBED/CodeDrops/beta/thread_example/mbed-client-example-6lowpan/yotta_modules/mbed-hal-ksdk-mcu -IC:/work/temp/MBED/CodeDrops/beta/thread_example/mbed-client-example-6lowpan/yotta_modules/uvisor-lib -IC:/work/temp/MBED/CodeDrops/beta/thread_example/mbed-client-example-6lowpan/yotta_modules/mbed-hal-k64f -IC:/work/temp/MBED/CodeDrops/beta/thread_example/mbed-client-example-6lowpan/yotta_modules/mbed-hal-frdm-k64f -IC:/work/temp/MBED/CodeDrops/beta/thread_example/mbed-client-example-6lowpan/yotta_modules/cmsis-core-freescale -IC:/work/temp/MBED/CodeDrops/beta/thread_example/mbed-client-example-6lowpan/yotta_modules/cmsis-core-k64f -IC:/work/temp/MBED/CodeDrops/beta/thread_example/mbed-client-example-6lowpan/yotta_modules/dlmalloc -IC:/work/temp/MBED/CodeDrops/beta/thread_example/mbed-client-example-6lowpan/yotta_modules/compiler-polyfill -IC:/work/temp/MBED/CodeDrops/beta/thread_example/mbed-client-example-6lowpan/yotta_modules/minar-platform -IC:/work/temp/MBED/CodeDrops/beta/thread_example/mbed-client-example-6lowpan/yotta_modules/minar-platform-mbed -IC:/work/temp/MBED/CodeDrops/beta/thread_example/mbed-client-example-6lowpan/yotta_modules/sal -IC:/work/temp/MBED/CodeDrops/beta/thread_example/mbed-client-example-6lowpan/yotta_modules/sal-stack-lwip -IC:/work/temp/MBED/CodeDrops/beta/thread_example/mbed-client-example-6lowpan/yotta_modules/sal-driver-lwip-k64f-eth -IC:/work/temp/MBED/CodeDrops/beta/thread_example/mbed-client-example-6lowpan/yotta_modules/sal-iface-eth -IC:/work/temp/MBED/CodeDrops/beta/thread_example/mbed-client-example-6lowpan/yotta_modules/mbedtls -IC:/work/temp/MBED/CodeDrops/beta/thread_example/mbed-client-example-6lowpan/yotta_modules/sal-stack-nanostack -IC:/work/temp/MBED/CodeDrops/beta/thread_example/mbed-client-example-6lowpan/yotta_modules/sal-stack-nanostack-eventloop -IC:/work/temp/MBED/CodeDrops/beta/thread_example/mbed-client-example-6lowpan/yotta_modules/nanostack-randlib -IC:/work/temp/MBED/CodeDrops/beta/thread_example/mbed-client-example-6lowpan/yotta_modules/mbed-6lowpan-eventloop-adaptor -IC:/work/temp/MBED/CodeDrops/beta/thread_example/mbed-client-example-6lowpan/yotta_modules/sal-iface-6lowpan -IC:/work/temp/MBED/CodeDrops/beta/thread_example/mbed-client-example-6lowpan/yotta_modules/sal-stack-lwip/mbed-net-lwip -IC:/work/temp/MBED/CodeDrops/beta/thread_example/mbed-client-example-6lowpan/yotta_modules/sal-stack-lwip/mbed-net-lwip/arch -IC:/work/temp/MBED/CodeDrops/beta/thread_example/mbed-client-example-6lowpan/yotta_modules/sal-stack-lwip/mbed-net-lwip/lwip -IC:/work/temp/MBED/CodeDrops/beta/thread_example/mbed-client-example-6lowpan/yotta_modules/sal-stack-lwip/mbed-net-lwip/lwip/include -IC:/work/temp/MBED/CodeDrops/beta/thread_example/mbed-client-example-6lowpan/yotta_modules/sal-stack-lwip/mbed-net-lwip/lwip/include/ipv4 -IC:/work/temp/MBED/CodeDrops/beta/thread_example/mbed-client-example-6lowpan/yotta_modules/sal-stack-lwip/mbed-net-lwip/lwip/include/netif -IC:/work/temp/MBED/CodeDrops/beta/thread_example/mbed-client-example-6lowpan/yotta_modules/sal-driver-lwip-k64f-eth/mbed-net-lwip-k64f -IC:/work/temp/MBED/CodeDrops/beta/thread_example/mbed-client-example-6lowpan/yotta_modules/sal-iface-eth/sal-iface-eth -IC:/work/temp/MBED/CodeDrops/beta/thread_example/mbed-client-example-6lowpan/yotta_modules/mbed-drivers/mbed -IC:/work/temp/MBED/CodeDrops/beta/thread_example/mbed-client-example-6lowpan/yotta_modules/mbed-hal/mbed-hal -IC:/work/temp/MBED/CodeDrops/beta/thread_example/mbed-client-example-6lowpan/yotta_modules/cmsis-core/cmsis-core -IC:/work/temp/MBED/CodeDrops/beta/thread_example/mbed-client-example-6lowpan/yotta_modules/mbed-hal-ksdk-mcu/mbed-hal-ksdk-mcu -IC:/work/temp/MBED/CodeDrops/beta/thread_example/mbed-client-example-6lowpan/yotta_modules/mbed-hal-ksdk-mcu/mbed-hal-ksdk-mcu/TARGET_KSDK_CODE/common/phyksz8081 -IC:/work/temp/MBED/CodeDrops/beta/thread_example/mbed-client-example-6lowpan/yotta_modules/mbed-hal-ksdk-mcu/mbed-hal-ksdk-mcu/TARGET_KSDK_CODE/drivers/clock -IC:/work/temp/MBED/CodeDrops/beta/thread_example/mbed-client-example-6lowpan/yotta_modules/mbed-hal-ksdk-mcu/mbed-hal-ksdk-mcu/TARGET_KSDK_CODE/drivers/enet -IC:/work/temp/MBED/CodeDrops/beta/thread_example/mbed-client-example-6lowpan/yotta_modules/mbed-hal-ksdk-mcu/mbed-hal-ksdk-mcu/TARGET_KSDK_CODE/drivers/interrupt -IC:/work/temp/MBED/CodeDrops/beta/thread_example/mbed-client-example-6lowpan/yotta_modules/mbed-hal-ksdk-mcu/mbed-hal-ksdk-mcu/TARGET_KSDK_CODE/drivers/pit -IC:/work/temp/MBED/CodeDrops/beta/thread_example/mbed-client-example-6lowpan/yotta_modules/mbed-hal-ksdk-mcu/mbed-hal-ksdk-mcu/TARGET_KSDK_CODE/drivers/pit/common -IC:/work/temp/MBED/CodeDrops/beta/thread_example/mbed-client-example-6lowpan/yotta_modules/mbed-hal-ksdk-mcu/mbed-hal-ksdk-mcu/TARGET_KSDK_CODE/hal/adc -IC:/work/temp/MBED/CodeDrops/beta/thread_example/mbed-client-example-6lowpan/yotta_modules/mbed-hal-ksdk-mcu/mbed-hal-ksdk-mcu/TARGET_KSDK_CODE/hal/can -IC:/work/temp/MBED/CodeDrops/beta/thread_example/mbed-client-example-6lowpan/yotta_modules/mbed-hal-ksdk-mcu/mbed-hal-ksdk-mcu/TARGET_KSDK_CODE/hal/dac -IC:/work/temp/MBED/CodeDrops/beta/thread_example/mbed-client-example-6lowpan/yotta_modules/mbed-hal-ksdk-mcu/mbed-hal-ksdk-mcu/TARGET_KSDK_CODE/hal/dmamux -IC:/work/temp/MBED/CodeDrops/beta/thread_example/mbed-client-example-6lowpan/yotta_modules/mbed-hal-ksdk-mcu/mbed-hal-ksdk-mcu/TARGET_KSDK_CODE/hal/dspi -IC:/work/temp/MBED/CodeDrops/beta/thread_example/mbed-client-example-6lowpan/yotta_modules/mbed-hal-ksdk-mcu/mbed-hal-ksdk-mcu/TARGET_KSDK_CODE/hal/edma -IC:/work/temp/MBED/CodeDrops/beta/thread_example/mbed-client-example-6lowpan/yotta_modules/mbed-hal-ksdk-mcu/mbed-hal-ksdk-mcu/TARGET_KSDK_CODE/hal/enet -IC:/work/temp/MBED/CodeDrops/beta/thread_example/mbed-client-example-6lowpan/yotta_modules/mbed-hal-ksdk-mcu/mbed-hal-ksdk-mcu/TARGET_KSDK_CODE/hal/flextimer -IC:/work/temp/MBED/CodeDrops/beta/thread_example/mbed-client-example-6lowpan/yotta_modules/mbed-hal-ksdk-mcu/mbed-hal-ksdk-mcu/TARGET_KSDK_CODE/hal/gpio -IC:/work/temp/MBED/CodeDrops/beta/thread_example/mbed-client-example-6lowpan/yotta_modules/mbed-hal-ksdk-mcu/mbed-hal [... cut ...] les/sockets/source/v0/TCPListener.cpp
 In file included from C:/work/temp/MBED/CodeDrops/beta/thread_example/mbed-client-example-6lowpan/yotta_modules/sockets/source/v0/TCPListener.cpp:18:0:

C:/work/temp/MBED/CodeDrops/beta/thread_example/mbed-client-example-6lowpan/yotta_modules/sockets/mbed-net-sockets/v0/TCPListener.h:22:39: fatal error: mbed-util/FunctionPointer.h: No such file or directory

#include "mbed-util/FunctionPointer.h"
                                   ^

compilation terminated.

artokin, 2015-09-21:
I'm sorry, application dependency chain is not up to date and that is why the build fails. Pull requests have been made to several repositories and hopefully we get this problem solved soon.

artokin, 2015-09-21:
Module dependencies are updated and the build should work now. @hwaldman, would you please verify in your environment?

hwaldman, 2015-09-21:
It builds now, thanks.

closed 2015-09-22 by artokin

markus-becker-tridonic-com, 2015-10-09:
The dependencies are not yet completely fixed (or broken again) in the yotta registry:

```
mbed-client-example-6lowpan/yotta_modules/sal-stack-nanostack-eventloop/source/event.cpp:28:39: fatal error: core-util/FunctionPointer.h: No such                                                                                file or directory
 #include "core-util/FunctionPointer.h"
                                       ^
compilation terminated.
```

This is the current dependency tree:
```
$ yotta ls
mbed-client-example-6lowpan 0.0.2
|_ mbed-client 0.2.20
| |_ mbed-client-c 0.1.10 yotta_modules\mbed-client-c
| | \_ nanostack-libservice 3.0.4 yotta_modules\nanostack-libservice
| |_ mbed-client-mbed-os 0.1.11 yotta_modules\mbed-client-mbed-os
| | |_ mbed-drivers 0.6.9 yotta_modules\mbed-drivers
| | | |_ mbed-hal 0.6.4 yotta_modules\mbed-hal
| | | | \_ mbed-hal-freescale 0.5.2 yotta_modules\mbed-hal-freescale
| | | |   \_ mbed-hal-ksdk-mcu 0.5.7 yotta_modules\mbed-hal-ksdk-mcu
| | | |     |_ uvisor-lib 0.7.25 yotta_modules\uvisor-lib
| | | |     \_ mbed-hal-k64f 0.3.6 yotta_modules\mbed-hal-k64f
| | | |       \_ mbed-hal-frdm-k64f 0.4.6 yotta_modules\mbed-hal-frdm-k64f
| | | |_ cmsis-core 0.2.7 yotta_modules\cmsis-core
| | | | \_ cmsis-core-freescale 0.1.4 yotta_modules\cmsis-core-freescale
| | | |   \_ cmsis-core-k64f 0.1.5 yotta_modules\cmsis-core-k64f
| | | |_ ualloc 0.0.10 yotta_modules\ualloc
| | | | \_ dlmalloc 0.0.6 yotta_modules\dlmalloc
| | | |_ minar 0.6.7 yotta_modules\minar
| | | | |_ compiler-polyfill 1.0.4 yotta_modules\compiler-polyfill
| | | | \_ minar-platform 0.3.4 yotta_modules\minar-platform
| | | |   \_ minar-platform-mbed 0.1.5 yotta_modules\minar-platform-mbed
| | | \_ core-util 0.0.16 yotta_modules\core-util
| | \_ sockets 0.3.6 yotta_modules\sockets
| |   \_ sal 0.2.3 yotta_modules\sal
| |     \_ sal-stack-lwip 0.2.8 yotta_modules\sal-stack-lwip
| |       |_ sal-driver-lwip-k64f-eth 0.1.3 yotta_modules\sal-driver-lwip-k64f-eth
| |       \_ sal-iface-eth 0.1.4 yotta_modules\sal-iface-eth
| \_ mbed-client-mbedtls 0.0.9 yotta_modules\mbed-client-mbedtls
|   \_ mbedtls 2.0.7 yotta_modules\mbedtls
|_ atmel-rf-driver 0.0.24
| \_ sal-stack-nanostack 3.0.0 yotta_modules\sal-stack-nanostack
|   |_ sal-stack-nanostack-eventloop 0.0.20 yotta_modules\sal-stack-nanostack-eventloop
|   | \_ mbed-6lowpan-eventloop-adaptor 0.0.23 yotta_modules\mbed-6lowpan-eventloop-adaptor
|   \_ nanostack-randlib 0.0.5 yotta_modules\nanostack-randlib
\_ mbed-mesh-api 0.3.5
  \_ sal-iface-6lowpan 0.2.14 yotta_modules\sal-iface-6lowpan
```

The dependency on core-util has to be >0.2.0, which seems to be satisfied by mbed-drivers >0.9.0, which in turn seems to be satisfied by mbed-client-mbed-os 1.0.7.