ARMmbed/mbed-client-example-6lowpan #26: Example fails to send packets

devran01, 2015-09-25:
Example fails to send packets and reports "send() not supported with SOCKET_DGRAM!".
Detailed log:
[INFO][m6LND]: Start 6LoWPAN ND Bootstrap
[DBG ][m6LND]: app_parse_network_event() 0
[INFO][m6LND]: 6LoWPAN ND bootstrap ready
[DBG ][m6LND]: ND Access Point: fd00:ff1:ce0b:a5e0:0:ff:fe00:e50e
[DBG ][m6LND]: ND Prefix 64: fd:00:0f:f1:ce:0b:a5:e0
[DBG ][m6LND]: GP IPv6: fd00:ff1:ce0b:a5e0:fec2:3d00:4:bbd0
[DBG ][m6LND]: MAC 16-bit: ff:ff
[DBG ][m6LND]: PAN ID: 06:91
[DBG ][m6LND]: MAC 64-bit: fc:c2:3d:00:00:04:bb:d0
[DBG ][m6LND]: IID (Based on MAC 64-bit address): fe:c2:3d:00:00:04:bb:d0
[DBG ][m6LND]: Channel: 1
[ERR ][ns_sal]: send() not supported with SOCKET_DGRAM!
[ERR ][ns_sal]: send() not supported with SOCKET_DGRAM!
[ERR ][ns_sal]: send() not supported with SOCKET_DGRAM!

artokin, 2015-09-25:
6LoWPAN doesn't support send() for UDP sockets. mbed-client-mbed-os has been updated to use send() method for data sending. Either we need to add send() support for 6LoWPAN IP stack or modify mbed-client-mbed-os to use send_to().

artokin, 2015-09-28:
This error is corrected by version `1.0.6` of https://github.com/ARMmbed/mbed-client-mbed-os.

closed 2015-09-28 by artokin