ARMmbed/mbed-client-example-6lowpan #28: Fail to build for FRDM-K64F

tjankow, 2015-10-13:
I'm trying to build the example (commit 193dff6) for FRDM-K64F and I'm facing following error:
```
In file included from /tmp/mbed-client-example-6lowpan/yotta_modules/sal/source/socket_abstract.c:17:0:
/tmp/mbed-client-example-6lowpan/yotta_modules/sal/mbed-net-socket-abstract/socket_api.h:95:19: fatal error: cmsis.h: No such file or directory
```
Outout of "yt list":
```
mbed-client-example-6lowpan 0.0.2
┣─ mbed-client 1.1.5
┃ ┣─ mbed-client-c 1.0.4 yotta_modules/mbed-client-c
┃ ┃ ┗─ nanostack-libservice 3.0.4 yotta_modules/nanostack-libservice
┃ ┣─ mbed-client-mbed-os 1.0.8 yotta_modules/mbed-client-mbed-os
┃ ┃ ┣─ mbed-drivers 0.10.2 yotta_modules/mbed-drivers
┃ ┃ ┃ ┣─ mbed-hal 0.8.1 yotta_modules/mbed-hal
┃ ┃ ┃ ┃ ┗─ mbed-hal-freescale 0.6.0 yotta_modules/mbed-hal-freescale
┃ ┃ ┃ ┃   ┗─ mbed-hal-ksdk-mcu 0.6.4 yotta_modules/mbed-hal-ksdk-mcu
┃ ┃ ┃ ┃     ┣─ uvisor-lib 0.8.4 yotta_modules/uvisor-lib
┃ ┃ ┃ ┃     ┗─ mbed-hal-k64f 0.4.3 yotta_modules/mbed-hal-k64f
┃ ┃ ┃ ┃       ┗─ mbed-hal-frdm-k64f 0.5.1 yotta_modules/mbed-hal-frdm-k64f
┃ ┃ ┃ ┣─ cmsis-core 0.4.2 yotta_modules/cmsis-core
┃ ┃ ┃ ┃ ┗─ cmsis-core-freescale 0.2.0 yotta_modules/cmsis-core-freescale
┃ ┃ ┃ ┃   ┗─ cmsis-core-k64f 0.2.0 yotta_modules/cmsis-core-k64f
┃ ┃ ┃ ┣─ ualloc 0.2.0 yotta_modules/ualloc
┃ ┃ ┃ ┃ ┗─ dlmalloc 0.0.6 yotta_modules/dlmalloc
┃ ┃ ┃ ┣─ minar 0.9.0 yotta_modules/minar
┃ ┃ ┃ ┃ ┗─ minar-platform 0.4.0 yotta_modules/minar-platform
┃ ┃ ┃ ┃   ┗─ minar-platform-mbed 0.2.2 yotta_modules/minar-platform-mbed
┃ ┃ ┃ ┣─ core-util 0.3.1 yotta_modules/core-util
┃ ┃ ┃ ┗─ compiler-polyfill 1.1.0 yotta_modules/compiler-polyfill
┃ ┃ ┗─ sockets 0.6.0 yotta_modules/sockets
┃ ┃   ┗─ sal 0.4.1 yotta_modules/sal
┃ ┗─ mbed-client-mbedtls 1.0.4 yotta_modules/mbed-client-mbedtls
┃   ┗─ mbedtls 2.1.3 yotta_modules/mbedtls
┣─ atmel-rf-driver 0.0.25
┃ ┗─ sal-stack-nanostack 3.0.0 yotta_modules/sal-stack-nanostack
┃   ┣─ sal-stack-nanostack-eventloop 0.0.21 yotta_modules/sal-stack-nanostack-eventloop
┃   ┃ ┗─ mbed-6lowpan-eventloop-adaptor 0.0.24 yotta_modules/mbed-6lowpan-eventloop-adaptor
┃   ┗─ nanostack-randlib 0.0.5 yotta_modules/nanostack-randlib
┗─ mbed-mesh-api 0.3.6
  ┗─ sal-iface-6lowpan 0.2.15 yotta_modules/sal-iface-6lowpan
```

I think this issue is strongly related to #23.

artokin, 2015-10-14:
The problem seems to happen only in windows environment, the example seems to work in ubuntu. Trying to investigate what is causing the problem.

tjankow, 2015-10-14:
Forgot to mention that earlier - I'm running on Ubuntu 14.04.

artokin, 2015-10-14:
It looks like ```sal``` is not pulling ```sal-stack-lwip ```. 

```yt ls``` in working ubuntu 14.04 environment is as follows:
```
mbed-client-example-6lowpan 0.0.2
┣─ mbed-client 1.1.5
┃ ┣─ mbed-client-c 1.0.4 yotta_modules/mbed-client-c
┃ ┃ ┗─ nanostack-libservice 3.0.4 yotta_modules/nanostack-libservice
┃ ┣─ mbed-client-mbed-os 1.0.8 yotta_modules/mbed-client-mbed-os
┃ ┃ ┣─ mbed-drivers 0.10.2 yotta_modules/mbed-drivers
┃ ┃ ┃ ┣─ mbed-hal 0.8.1 yotta_modules/mbed-hal
┃ ┃ ┃ ┃ ┗─ mbed-hal-freescale 0.6.0 yotta_modules/mbed-hal-freescale
┃ ┃ ┃ ┃   ┗─ mbed-hal-ksdk-mcu 0.6.4 yotta_modules/mbed-hal-ksdk-mcu
┃ ┃ ┃ ┃     ┣─ uvisor-lib 0.8.4 yotta_modules/uvisor-lib
┃ ┃ ┃ ┃     ┗─ mbed-hal-k64f 0.4.4 yotta_modules/mbed-hal-k64f
┃ ┃ ┃ ┃       ┗─ mbed-hal-frdm-k64f 0.5.1 yotta_modules/mbed-hal-frdm-k64f
┃ ┃ ┃ ┣─ cmsis-core 0.4.2 yotta_modules/cmsis-core
┃ ┃ ┃ ┃ ┗─ cmsis-core-freescale 0.2.0 yotta_modules/cmsis-core-freescale
┃ ┃ ┃ ┃   ┗─ cmsis-core-k64f 0.2.0 yotta_modules/cmsis-core-k64f
┃ ┃ ┃ ┣─ ualloc 0.2.0 yotta_modules/ualloc
┃ ┃ ┃ ┃ ┗─ dlmalloc 0.0.6 yotta_modules/dlmalloc
┃ ┃ ┃ ┣─ minar 0.9.0 yotta_modules/minar
┃ ┃ ┃ ┃ ┗─ minar-platform 0.4.0 yotta_modules/minar-platform
┃ ┃ ┃ ┃   ┗─ minar-platform-mbed 0.2.2 yotta_modules/minar-platform-mbed
┃ ┃ ┃ ┣─ core-util 0.3.1 yotta_modules/core-util
┃ ┃ ┃ ┗─ compiler-polyfill 1.1.0 yotta_modules/compiler-polyfill
┃ ┃ ┗─ sockets 0.6.0 yotta_modules/sockets
┃ ┃   ┗─ sal 0.4.1 yotta_modules/sal
┃ ┃     ┗─ sal-stack-lwip 0.4.0 yotta_modules/sal-stack-lwip
┃ ┃       ┣─ sal-driver-lwip-k64f-eth 0.2.0 yotta_modules/sal-driver-lwip-k64f-eth
┃ ┃       ┗─ sal-iface-eth 0.3.0 yotta_modules/sal-iface-eth
┃ ┗─ mbed-client-mbedtls 1.0.4 yotta_modules/mbed-client-mbedtls
┃   ┗─ mbedtls 2.1.3 yotta_modules/mbedtls
┣─ atmel-rf-driver 0.0.25
┃ ┗─ sal-stack-nanostack 3.0.0 yotta_modules/sal-stack-nanostack
┃   ┣─ sal-stack-nanostack-eventloop 0.0.21 yotta_modules/sal-stack-nanostack-eventloop
┃   ┃ ┗─ mbed-6lowpan-eventloop-adaptor 0.0.24 yotta_modules/mbed-6lowpan-eventloop-adaptor
┃   ┗─ nanostack-randlib 0.0.5 yotta_modules/nanostack-randlib
┗─ mbed-mesh-api 0.3.6
  ┗─ sal-iface-6lowpan 0.2.15 yotta_modules/sal-iface-6lowpan
```

```yt ls``` in windows where problem occurs:
```
mbed-client-example-6lowpan 0.0.2
|_ mbed-client 1.1.5
| |_ mbed-client-c 1.0.4 yotta_modules\mbed-client-c
| | \_ nanostack-libservice 3.0.4 yotta_modules\nanostack-libservice
| |_ mbed-client-mbed-os 1.0.8 yotta_modules\mbed-client-mbed-os
| | |_ mbed-drivers 0.10.2 yotta_modules\mbed-drivers
| | | |_ mbed-hal 0.8.1 yotta_modules\mbed-hal
| | | | \_ mbed-hal-freescale 0.6.0 yotta_modules\mbed-hal-freescale
| | | |   \_ mbed-hal-ksdk-mcu 0.6.4 yotta_modules\mbed-hal-ksdk-mcu
| | | |     |_ uvisor-lib 0.8.4 yotta_modules\uvisor-lib
| | | |     \_ mbed-hal-k64f 0.4.3 yotta_modules\mbed-hal-k64f
| | | |       \_ mbed-hal-frdm-k64f 0.5.1 yotta_modules\mbed-hal-frdm-k64f
| | | |_ cmsis-core 0.4.2 yotta_modules\cmsis-core
| | | | \_ cmsis-core-freescale 0.2.0 yotta_modules\cmsis-core-freescale
| | | |   \_ cmsis-core-k64f 0.2.0 yotta_modules\cmsis-core-k64f
| | | |_ ualloc 0.2.0 yotta_modules\ualloc
| | | | \_ dlmalloc 0.0.6 yotta_modules\dlmalloc
| | | |_ minar 0.9.0 yotta_modules\minar
| | | | \_ minar-platform 0.4.0 yotta_modules\minar-platform
| | | |   \_ minar-platform-mbed 0.2.2 yotta_modules\minar-platform-mbed
| | | |_ core-util 0.3.1 yotta_modules\core-util
| | | \_ compiler-polyfill 1.1.0 yotta_modules\compiler-polyfill
| | \_ sockets 0.6.0 yotta_modules\sockets
| |   \_ sal 0.4.1 yotta_modules\sal
| \_ mbed-client-mbedtls 1.0.4 yotta_modules\mbed-client-mbedtls
|   \_ mbedtls 2.1.3 yotta_modules\mbedtls
|_ atmel-rf-driver 0.0.25
| \_ sal-stack-nanostack 3.0.0 yotta_modules\sal-stack-nanostack
|   |_ sal-stack-nanostack-eventloop 0.0.21 yotta_modules\sal-stack-nanostack-ev
entloop
|   | \_ mbed-6lowpan-eventloop-adaptor 0.0.24 yotta_modules\mbed-6lowpan-eventl
oop-adaptor
|   \_ nanostack-randlib 0.0.5 yotta_modules\nanostack-randlib
\_ mbed-mesh-api 0.3.6
  \_ sal-iface-6lowpan 0.2.15 yotta_modules\sal-iface-6lowpan
```

@bremoran , would you please help in this issue?

tjankow, 2015-10-14:
I checked one more time my environment , because It sounded somehow weird that yotta wasn't downloading some components. Apparently I had outdated yotta installed, updating it to v0.8.4 solved the problem.

artokin, 2015-10-14:
Excellent, updating yotta solved the problem in windows environment as well.

closed 2015-10-14 by artokin