ARMmbed/mbed-client-example-6lowpan #31: Error prints / not able to read resources with device connector

ErkkiMehtala, 2015-10-26:
Here some errors whic came when testing with instructions from Readme.txt (1GHz radios, 6lowban)
After reset, in beginning was error 10

In https://connector-test-sl.dev.mbed.com/
I tried to read resources form Device connnector API console quite many times. 
I was able to read endpoint itseld and it showed list of resources. 
But any deeper request to resources returned http errors 402 or 404
all the time. For example these did not work:

https://ds-test-sl.dev.mbed.com/endpoints/c6881b0d-3873-4ef3-a747-3318ea5f63c4/3/0/1?sync
https://ds-test-sl.dev.mbed.com/endpoints/c6881b0d-3873-4ef3-a747-3318ea5f63c4/Test/0/D

Quite many error prints, but don't know from which operation exactly those came.

After reset:
-unregister with button SW3 which seemed to work and led changed color
-increased counter several times by pressing button SW2, print with no errors 
-API calls from connector like earlier, just endpoint itself able to read
-no error prints any more

Log:

[ERR ][mbedclient]: error() 10
[DBG ][m6La]: init()
[DBG ][m6La]: connect()
[INFO][m6LND]: Start 6LoWPAN ND Bootstrap
[DBG ][m6LND]: app_parse_network_event() 0
[INFO][m6LND]: 6LoWPAN ND bootstrap ready
[DBG ][m6LND]: ND Access Point: 2001:14b8:100:83be:0:ff:fe00:a9c0
[DBG ][m6LND]: ND Prefix 64: 20:01:14:b8:01:00:83:be
[DBG ][m6LND]: GP IPv6: 2001:14b8:100:83be:fec2:3d00:4:b2c7
[DBG ][m6LND]: MAC 16-bit: ff:ff
[DBG ][m6LND]: PAN ID: 06:91
[DBG ][m6LND]: MAC 64-bit: fc:c2:3d:00:00:04:b2:c7
[DBG ][m6LND]: IID (Based on MAC 64-bit address): fe:c2:3d:00:00:04:b2:c7
[DBG ][m6LND]: Channel: 12
[DBG ][mbedclient]: mesh_network_handler() 0
[DBG ][mbedclient]: waiting 15s before sending registration...
[DBG ][mbedclient]: send_registration()
[DBG ][ns_wrap]: SOCKET_TX_DONE, 163 bytes sent
[DBG ][ns_wrap]: SOCKET_DATA, sock=1, bytes=60
[DBG ][ns_wrap]: SOCKET_DATA, sock=1, bytes=60
[DBG ][ns_wrap]: SOCKET_TX_DONE, 195 bytes sent
[DBG ][ns_wrap]: SOCKET_DATA, sock=1, bytes=112
[DBG ][ns_wrap]: SOCKET_DATA, sock=1, bytes=384
[DBG ][ns_wrap]: SOCKET_DATA, sock=1, bytes=201
[DBG ][ns_wrap]: SOCKET_DATA, sock=1, bytes=178
[DBG ][ns_wrap]: SOCKET_DATA, sock=1, bytes=25
[DBG ][ns_wrap]: SOCKET_TX_DONE, 472 bytes sent
[DBG ][ns_wrap]: SOCKET_TX_DONE, 123 bytes sent
[DBG ][ns_wrap]: SOCKET_TX_DONE, 100 bytes sent
[DBG ][ns_wrap]: SOCKET_TX_DONE, 14 bytes sent
[DBG ][ns_wrap]: SOCKET_TX_DONE, 61 bytes sent
[DBG ][ns_wrap]: SOCKET_DATA, sock=1, bytes=112
[DBG ][ns_wrap]: SOCKET_DATA, sock=1, bytes=384
[DBG ][ns_wrap]: SOCKET_DATA, sock=1, bytes=201
[DBG ][ns_wrap]: SOCKET_DATA, sock=1, bytes=178
[DBG ][ns_wrap]: SOCKET_DATA, sock=1, bytes=25
[DBG ][ns_wrap]: SOCKET_DATA, sock=1, bytes=14
[DBG ][ns_wrap]: SOCKET_DATA, sock=1, bytes=61
[DBG ][ns_wrap]: SOCKET_TX_DONE, 472 bytes sent
[DBG ][ns_wrap]: SOCKET_TX_DONE, 123 bytes sent
[DBG ][ns_wrap]: SOCKET_TX_DONE, 100 bytes sent
[DBG ][ns_wrap]: SOCKET_TX_DONE, 14 bytes sent
[DBG ][ns_wrap]: SOCKET_TX_DONE, 61 bytes sent
[DBG ][ns_wrap]: SOCKET_TX_DONE, 258 bytes sent
[DBG ][ns_wrap]: SOCKET_DATA, sock=1, bytes=14
[DBG ][ns_wrap]: SOCKET_DATA, sock=1, bytes=61
[DBG ][ns_wrap]: SOCKET_DATA, sock=1, bytes=61
[DBG ][ns_wrap]: SOCKET_DATA, sock=1, bytes=93
[... 205 lines cut ...]
[DBG ][m6LND]: app_parse_network_event() 3
[DBG ][m6LND]: Link Layer Scan Fail: No Beacons
[DBG ][m6LND]: Restart bootstrap
[DBG ][m6LND]: app_parse_network_event() 3
[DBG ][m6LND]: Link Layer Scan Fail: No Beacons
[DBG ][m6LND]: Restart bootstrap
[DBG ][m6LND]: app_parse_network_event() 0
[INFO][m6LND]: 6LoWPAN ND bootstrap ready
[DBG ][m6LND]: ND Access Point: 2001:14b8:100:83be:0:ff:fe00:a9c0
[DBG ][m6LND]: ND Prefix 64: 20:01:14:b8:01:00:83:be
[DBG ][m6LND]: GP IPv6: 2001:14b8:100:83be:fec2:3d00:4:b2c7
[DBG ][m6LND]: MAC 16-bit: ff:ff
[DBG ][m6LND]: PAN ID: 06:91
[DBG ][m6LND]: MAC 64-bit: fc:c2:3d:00:00:04:b2:c7
[DBG ][m6LND]: IID (Based on MAC 64-bit address): fe:c2:3d:00:00:04:b2:c7
[DBG ][m6LND]: Channel: 12
[DBG ][mbedclient]: mesh_network_handler() 0
[DBG ][mbedclient]: waiting 15s before sending registration...
[DBG ][mbedclient]: send_registration()
[DBG ][ns_wrap]: SOCKET_TX_DONE, 163 bytes sent
[DBG ][mbedclient]: update_resource() 1
[DBG ][mbedclient]: update_resource() 2
[DBG ][mbedclient]: update_resource() 3
[DBG ][mbedclient]: update_resource() 4
[DBG ][mbedclient]: update_resource() 5

rainierwolfcastle, 2015-10-26:
ARM Internal Ref: ONME-2018

ErkkiMehtala, 2015-10-27:
Testing now with tread 2.4GHz. First all worked fine, using buttons and asking values from server.
After a while, when asking resources from server, there started to come error 10 again 
(like was situation in earlier 6lowban test). Just this time timeout error came mostly.

I was able to read endpoint, but not any resources inside of it.

[DBG ][ns_sal]: ns_sal_socket_send_to()
[DBG ][ns_wrap]: ns_wrapper_socket_send_to: sock_id=5, length=258, port=5684
[DBG ][ns_wrap]: socket_callback() sock=5, event=144, interface=0, data len=258
[DBG ][ns_wrap]: SOCKET_TX_DONE, 258 bytes sent
[DBG ][ns_wrap]: socket_callback() sock=5, event=0, interface=0, data len=93
[DBG ][ns_wrap]: SOCKET_DATA, sock=5, bytes=93
[DBG ][ns_sal]: ns_sal_socket_send_to()
[DBG ][ns_wrap]: ns_wrapper_socket_send_to: sock_id=5, length=258, port=5684
[DBG ][ns_wrap]: socket_callback() sock=5, event=144, interface=0, data len=258
[DBG ][ns_wrap]: SOCKET_TX_DONE, 258 bytes sent
[DBG ][ns_wrap]: socket_callback() sock=5, event=0, interface=0, data len=93
[DBG ][ns_wrap]: SOCKET_DATA, sock=5, bytes=93
[DBG ][ns_sal]: ns_sal_socket_send_to()
[DBG ][ns_wrap]: ns_wrapper_socket_send_to: sock_id=5, length=101, port=5684
[DBG ][ns_wrap]: socket_callback() sock=5, event=144, interface=0, data len=101
[DBG ][ns_wrap]: SOCKET_TX_DONE, 101 bytes sent
[DBG ][ns_wrap]: socket_callback() sock=5, event=0, interface=0, data len=45
[DBG ][ns_wrap]: SOCKET_DATA, sock=5, bytes=45
[DBG ][ns_sal]: ns_sal_socket_send_to()
[DBG ][ns_wrap]: ns_wrapper_socket_send_to: sock_id=5, length=101, port=5684
[DBG ][ns_wrap]: socket_callback() sock=5, event=144, interface=0, data len=101
[DBG ][ns_wrap]: SOCKET_TX_DONE, 101 bytes sent
[DBG ][ns_wrap]: socket_callback() sock=5, event=0, interface=0, data len=45
[DBG ][ns_wrap]: SOCKET_DATA, sock=5, bytes=45
[DBG ][ns_sal]: ns_sal_socket_send_to()
[DBG ][ns_wrap]: ns_wrapper_socket_send_to: sock_id=5, length=41, port=5684
[DBG ][ns_wrap]: socket_callback() sock=5, event=144, interface=0, data len=41
[DBG ][ns_wrap]: SOCKET_TX_DONE, 41 bytes sent
[DBG ][ns_sal]: ns_sal_socket_send_to()
[DBG ][ns_wrap]: ns_wrapper_socket_send_to: sock_id=5, length=101, port=5684
[DBG ][ns_wrap]: socket_callback() sock=5, event=144, interface=0, data len=101
[DBG ][ns_wrap]: SOCKET_TX_DONE, 101 bytes sent
[DBG ][ns_wrap]: socket_callback() sock=5, event=0, interface=0, data len=45
[DBG ][ns_wrap]: SOCKET_DATA, sock=5, bytes=45
[DBG ][ns_sal]: ns_sal_socket_send_to()
[DBG ][ns_wrap]: ns_wrapper_socket_send_to: sock_id=5, length=101, port=5684
[DBG ][ns_wrap]: socket_callback() sock=5, event=144, interface=0, data len=101
[DBG ][ns_wrap]: SOCKET_TX_DONE, 101 bytes sent
[DBG ][ns_wrap]: socket_callback() sock=5, event=0, interface=0, data len=45
[DBG ][ns_wrap]: SOCKET_DATA, sock=5, bytes=45
[DBG ][ns_sal]: ns_sal_socket_send_to()
[DBG ][ns_wrap]: ns_wrapper_socket_send_to: sock_id=5, length=101, port=5684
[DBG ][ns_wrap]: socket_callback() sock=5, event=144, interface=0, data len=101
[DBG ][ns_wrap]: SOCKET_TX_DONE, 101 bytes sent
[DBG ][ns_wrap]: socket_callback() sock=5, event=0, interface=0, data len=45
[DBG ][ns_wrap]: SOCKET_DATA, sock=5, bytes=45
[DBG ][ns_sal]: ns_sal_socket_send_to()
[DBG ][ns_wrap]: ns_wrapper_socket_send_to: sock_id=5, length=41, port=5684
[DBG ][ns_wrap]: socket_callback() sock=5, event=144, interface=0, data len=41
[DBG ][ns_wrap]: SOCKET_TX_DONE, 41 bytes sent
[DBG ][ns_sal]: ns_sal_socket_send_to()
[DBG ][ns_wrap]: ns_wrapper_socket_send_to: sock_id=5, length=101, port=5684
[DBG ][ns_wrap]: socket_callback() sock=5, event=144, interface=0, data len=101
[DBG ][ns_wrap]: SOCKET_TX_DONE, 101 bytes sent
[DBG ][ns_wrap]: socket_callback() sock=5, event=0, interface=0, data len=45
[DBG ][ns_wrap]: SOCKET_DATA, sock=5, bytes=45
[DBG ][ns_sal]: ns_sal_socket_send_to()
[DBG ][ns_wrap]: ns_wrapper_socket_send_to: sock_id=5, length=101, port=5684
[DBG ][ns_wrap]: socket_callback() sock=5, event=144, interface=0, data len=101
[DBG ][ns_wrap]: SOCKET_TX_DONE, 101 bytes sent
[DBG ][ns_wrap]: socket_callback() sock=5, event=0, interface=0, data len=45
[DBG ][ns_wrap]: SOCKET_DATA, sock=5, bytes=45
[DBG ][ns_sal]: ns_sal_socket_send_to()
[DBG ][ns_wrap]: ns_wrapper_socket_send_to: sock_id=5, length=101, port=5684
[... 612 lines cut ...]
[DBG ][mbedclient]: send_registration()
[DBG ][ns_sal]: ns_sal_socket_send_to()
[DBG ][ns_wrap]: ns_wrapper_socket_send_to: sock_id=7, length=163, port=5684
[ERR ][ns_sal]: ns_sal_socket_send_to: error=-4
[ERR ][mbedclient]: error() 6
[DBG ][m6La]: init()
[DBG ][m6La]: connect()
[INFO][m6Thread]: Start 6LoWPAN Thread bootstrap
[DBG ][m6Thread]: app_parse_network_event() 0
[INFO][m6Thread]: Thread bootstrap ready
[DBG ][m6Thread]: MAC 16-bit: 00:00
[DBG ][m6Thread]: PAN ID: fa:ce
[DBG ][m6Thread]: MAC 64-bit: fe:c2:3d:00:00:04:b2:c7
[DBG ][m6Thread]: IID (Based on MAC 64-bit address): fc:c2:3d:00:00:04:b2:c7
[DBG ][mbedclient]: mesh_network_handler() 0
[DBG ][ns_sal]: ns_sal_socket_create() af=2 pf=1
[DBG ][ns_wrap]: ns_wrapper_socket_open()
[DBG ][ns_sal]: ns_sal_str2addr() 0:0:0:0:0:0:0:0
[DBG ][ns_wrap]: ns_wrapper_socket_bind() sock=7
[DBG ][mbedclient]: waiting 15s before sending registration...
[DBG ][mbedclient]: send_registration()
[DBG ][ns_sal]: ns_sal_socket_send_to()
[DBG ][ns_wrap]: ns_wrapper_socket_send_to: sock_id=7, length=163, port=5684
[ERR ][ns_sal]: ns_sal_socket_send_to: error=-4
[ERR ][mbedclient]: error() 6

SeppoTakalo, 2015-10-27:
Fixed error handling in https://github.com/ARMmbed/mbed-client-example-6lowpan/commits/TechCon-release

closed 2015-10-27 by SeppoTakalo