ARMmbed/mbed-client-example-6lowpan #35: Does not work with ARMCC in Thread mode.

SeppoTakalo, 2015-10-29:
If I compile the example application for target frdm-k64f-gcc, everything works.

If I compile the application with target frdm-k64f-armcc, I see that the node does not join the network.

I see output like this from serial port:
~~~~
[DBG ][m6La]: init()
[DBG ][m6La]: connect()
[INFO][m6Thread]: Start 6LoWPAN Thread bootstrap
~~~~
Then the bootstrap seems to take forewer to complete.

After bootstrap: (2minutes later)
~~~~
[DBG ][m6Thread]: app_parse_network_event() 0
[INFO][m6Thread]: Thread bootstrap ready
[DBG ][m6Thread]: MAC 16-bit: 00:01
[DBG ][m6Thread]: PAN ID: fa:ce
[DBG ][m6Thread]: MAC 64-bit: fe:c2:3d:00:00:04:99:99
[DBG ][m6Thread]: IID (Based on MAC 64-bit address): fc:c2:3d:00:00:04:99:99
[DBG ][mbedclient]: mesh_network_handler() 0
[DBG ][mbedclient]: waiting 5s before sending registration...
[DBG ][mbedclient]: send_registration()
[ERR ][ns_sal]: ns_sal_socket_send_to: error=-4
~~~~

I can send Wireshark capture of the event, seems that I cannot attach it here.

rainierwolfcastle, 2015-10-30:
ARM Internal Ref: ONME-2074

SeppoTakalo, 2015-11-03:
Failed using ARMCC 5.03 in Linux.
Works when builded using ARMCC 5.05 in Windows.

SeppoTakalo, 2015-11-03:
Verified also that ARMCC 5.03 on Windows produced same kind of error.

Build works on all platforms, but ARMCC 5.03 ones are not able to join the network.

artokin, 2015-11-03:
Verified that ARMCC 5.03 (Ubuntu 14.04) is working fine with 6LoWPAN ND mode.

SeppoTakalo, 2016-03-22:
ARMCC 5.03 is not supported. Due the C++11 support, ARMCC 5.05 is minimal requirement.

closed 2016-03-22 by SeppoTakalo