ARMmbed/mbed-client-example-6lowpan #36: Reconnection fails after using example for ~12 minutes

artokin, 2015-10-30:
Example application is working fine about 12 minutes but then the device is trying to reconnect again but reconnection fails.

The log snipplet after using example about 12 minutes:
[DBG ][mbedclient]: update_resource() 56
[DBG ][ns_wrap]: SOCKET_DATA, sock=1, bytes=50
[DBG ][ns_wrap]: SOCKET_TX_DONE, 48 bytes sent
[DBG ][ns_wrap]: SOCKET_TX_DONE, 97 bytes sent
[DBG ][ns_wrap]: SOCKET_DATA, sock=1, bytes=45
[DBG ][mbedclient]: Registration updated
[DBG ][ns_wrap]: SOCKET_TX_DONE, 41 bytes sent
[DBG ][ns_wrap]: SOCKET_TX_DONE, 97 bytes sent
[DBG ][ns_wrap]: SOCKET_DATA, sock=1, bytes=45
[DBG ][ns_wrap]: SOCKET_TX_DONE, 97 bytes sent
[DBG ][ns_wrap]: SOCKET_DATA, sock=1, bytes=45
[DBG ][ns_wrap]: SOCKET_TX_DONE, 97 bytes sent
[DBG ][ns_wrap]: SOCKET_DATA, sock=1, bytes=45
[DBG ][mbedclient]: Registration updated
[DBG ][ns_wrap]: SOCKET_TX_DONE, 97 bytes sent
[DBG ][ns_wrap]: SOCKET_DATA, sock=1, bytes=45
[DBG ][ns_wrap]: SOCKET_TX_DONE, 97 bytes sent
[DBG ][ns_wrap]: SOCKET_DATA, sock=1, bytes=45
[DBG ][mbedclient]: Registration updated
[DBG ][ns_wrap]: SOCKET_TX_DONE, 41 bytes sent
[DBG ][ns_wrap]: SOCKET_TX_DONE, 97 bytes sent
[DBG ][ns_wrap]: SOCKET_DATA, sock=1, bytes=45
[DBG ][ns_wrap]: SOCKET_TX_DONE, 97 bytes sent
[DBG ][ns_wrap]: SOCKET_DATA, sock=1, bytes=45
[DBG ][ns_wrap]: SOCKET_TX_DONE, 97 bytes sent
[DBG ][ns_wrap]: SOCKET_DATA, sock=1, bytes=45
[DBG ][mbedclient]: Registration updated
[DBG ][ns_wrap]: SOCKET_TX_DONE, 41 bytes sent
[DBG ][ns_wrap]: SOCKET_TX_DONE, 97 bytes sent
[DBG ][ns_wrap]: SOCKET_DATA, sock=1, bytes=45
[DBG ][ns_wrap]: SOCKET_TX_DONE, 97 bytes sent
[DBG ][ns_wrap]: SOCKET_DATA, sock=1, bytes=45
[DBG ][mbedclient]: Registration updated
[DBG ][ns_wrap]: SOCKET_TX_DONE, 41 bytes sent
[DBG ][ns_wrap]: SOCKET_TX_DONE, 97 bytes sent
[DBG ][ns_wrap]: SOCKET_DATA, sock=1, bytes=45
[DBG ][ns_wrap]: SOCKET_TX_DONE, 97 bytes sent
[DBG ][ns_wrap]: SOCKET_DATA, sock=1, bytes=45
[DBG ][ns_wrap]: SOCKET_TX_DONE, 97 bytes sent
[DBG ][ns_wrap]: SOCKET_DATA, sock=1, bytes=45
[DBG ][mbedclient]: Registration updated
[DBG ][ns_wrap]: SOCKET_TX_DONE, 97 bytes sent
[DBG ][ns_wrap]: SOCKET_DATA, sock=1, bytes=45
[ERR ][mbedclient]: error() 10
[DBG ][mbedclient]: Reconnecting to server
[DBG ][mbedclient]: waiting 5s before sending registration...
[DBG ][mbedclient]: send_registration()
[DBG ][ns_wrap]: SOCKET_TX_DONE, 163 bytes sent
[DBG ][ns_wrap]: SOCKET_DATA, sock=2, bytes=60
[DBG ][ns_wrap]: SOCKET_TX_DONE, 195 bytes sent
[DBG ][ns_wrap]: SOCKET_DATA, sock=2, bytes=112
[DBG ][mbedclient]: waiting 5s before sending registration...
[DBG ][mbedclient]: send_registration()
[DBG ][ns_wrap]: SOCKET_TX_DONE, 163 bytes sent
[DBG ][ns_wrap]: SOCKET_DATA, sock=3, bytes=60
[DBG ][ns_wrap]: SOCKET_TX_DONE, 195 bytes sent
[DBG ][ns_wrap]: SOCKET_DATA, sock=3, bytes=112
[DBG ][ns_wrap]: SOCKET_DATA, sock=3, bytes=384
[DBG ][ns_wrap]: SOCKET_DATA, sock=3, bytes=202
[DBG ][ns_wrap]: SOCKET_DATA, sock=3, bytes=178
[DBG ][ns_wrap]: SOCKET_DATA, sock=3, bytes=25
[DBG ][ns_wrap]: SOCKET_TX_DONE, 468 bytes sent
[DBG ][ns_wrap]: SOCKET_TX_DONE, 123 bytes sent
[DBG ][ns_wrap]: SOCKET_TX_DONE, 100 bytes sent
[DBG ][ns_wrap]: SOCKET_TX_DONE, 14 bytes sent
[DBG ][ns_wrap]: SOCKET_TX_DONE, 61 bytes sent
[DBG ][ns_wrap]: SOCKET_DATA, sock=3, bytes=112
[DBG ][ns_wrap]: SOCKET_DATA, sock=3, bytes=384
[DBG ][ns_wrap]: SOCKET_DATA, sock=3, bytes=202
[DBG ][ns_wrap]: SOCKET_DATA, sock=3, bytes=14
[DBG ][mbedclient]: waiting 5s before sending registration...
[DBG ][mbedclient]: send_registration()
[ERR ][mbedclient]: error() 6
[DBG ][mbedclient]: Reconnecting to server
[DBG ][mbedclient]: waiting 5s before sending registration...
[DBG ][mbedclient]: send_registration()
[ERR ][mbedclient]: error() 6
[DBG ][mbedclient]: Reconnecting to server
[DBG ][mbedclient]: waiting 5s before sending registration...
[DBG ][mbedclient]: send_registration()
[ERR ][mbedclient]: error() 6
[DBG ][mbedclient]: Reconnecting to server
[DBG ][mbedclient]: waiting 5s before sending registration...
[DBG ][mbedclient]: send_registration()

rainierwolfcastle, 2015-10-30:
ARM Internal Ref: ONME-2073

SeppoTakalo, 2015-11-03:
Fixed

closed 2015-11-03 by SeppoTakalo