ARMmbed/mbed-client-example-6lowpan #37: Reconnections fail after 3 attempts.

SeppoTakalo, 2015-10-30:
When running the application on 6LoWPAN network it re-creates the interface classes after each network failures, this causes the re-connections to start failing after 3 re-connections.

After this, I will constantly see:
~~~
[DBG ][mbedclient]: waiting 5s before sending registration...
[DBG ][mbedclient]: send_registration()
[ERR ][mbedclient]: error() 6
~~~

How to re-produce:
1. Modify application so that when receiving object_registered callback, you destroy the connection and restart the registration proses.
2. Run application.
3. Observe that after 3 succesful registrations, it starts to fail.

SeppoTakalo, 2015-10-30:
CC: @yogpan01 @artokin @juhhei01

yogpan01, 2015-10-30:
mbed client is using mbed OS socket and when any application deletes mbed client instance, it will call mbedSocket destructor, it is the responsibility of underlying porting layer to ensure that proper cleanup happens.

SeppoTakalo, 2015-10-30:
Yes, of course. I just CC:d that to you also because you might be interested, and you should be familiar with this issue.
We continue the investigation to find to root cause for this.

rainierwolfcastle, 2015-10-30:
ARM Internal Ref: ONME-2072

artokin, 2015-10-30:
@AnttiKauppila , would you please help on this issue? It looks like there is something wrong in SSL handshake, in M2MConnectionHandlerPimpl::dns_handler, start_connecting_non_blocking() is returning error.

AnttiKauppila, 2015-11-02:
Fixed in mbed-client-mbedtls version 1.0.8

closed 2015-11-02 by SeppoTakalo