ARMmbed/mbed-client-example-6lowpan #38: No BR FW available for static configuration

mlnx, 2015-10-30:
Using TechCon-release.

The GW_Binary folder has two files:
- gateway6LoWPANDynamic.bin
- gatewayThreadDynamic.bin

There are no BR fw files that are required to test this locally

hasnainvirk, 2015-10-30:
Local testing is not required in TechCon release OOB. A connection to Device Connector is required, that's why only Dynamic binaries are included.

mlnx, 2015-10-30:
The TechCon release branch is not only used for OOB, but also to generate a tarball to be available to many people.
Please can we have the BR static FW included in this branch?
@SeppoTakalo @bogdanm 
Thanks

rainierwolfcastle, 2015-11-02:
ARM Internal Ref: ONME-2078

artokin, 2015-11-18:
Static binaries merged to master in commit 3abb33a4eb060df56b55b79b3e972c57af56636d.

closed 2015-11-18 by artokin