ARMmbed/mbed-client-example-6lowpan #39: 6lowpan and Thread examples don't work any more

hasnainvirk, 2015-10-30:
Setup: Linux, Windows both
Compilers: gcc, armcc both

I am receiving Network Error with all of our examples. I couldn't figure out the reason. Might be something to do with mbed-tls connection handshakes or some sort of device-connector issue

artokin, 2015-11-02:
There is something wrong in security, seems to work fine with "M2MSecurity::NoSecurity" and port 5683.

SeppoTakalo, 2015-11-02:
~~~~
[DBG ][mbedclient]: send_registration()
[DBG ][ns_wrap]: SOCKET_TX_DONE, 163 bytes sent
[DBG ][ns_wrap]: SOCKET_TX_DONE, 163 bytes sent
[DBG ][ns_wrap]: SOCKET_TX_DONE, 163 bytes sent
[DBG ][ns_wrap]: SOCKET_TX_DONE, 163 bytes sent
[DBG ][ns_wrap]: SOCKET_DATA, sock=5, bytes=60
[DBG ][ns_wrap]: SOCKET_TX_DONE, 195 bytes sent
[DBG ][ns_wrap]: SOCKET_DATA, sock=5, bytes=112
[DBG ][ns_wrap]: SOCKET_DATA, sock=5, bytes=384
[DBG ][ns_wrap]: SOCKET_DATA, sock=5, bytes=202
[DBG ][ns_wrap]: SOCKET_DATA, sock=5, bytes=178
[DBG ][ns_wrap]: SOCKET_DATA, sock=5, bytes=25
[DBG ][ns_wrap]: SOCKET_TX_DONE, 195 bytes sent
[DBG ][ns_wrap]: SOCKET_DATA, sock=5, bytes=112
[DBG ][ns_wrap]: SOCKET_DATA, sock=5, bytes=384
[DBG ][ns_wrap]: SOCKET_DATA, sock=5, bytes=202
[DBG ][ns_wrap]: SOCKET_DATA, sock=5, bytes=178
[DBG ][ns_wrap]: SOCKET_DATA, sock=5, bytes=25
[DBG ][ns_wrap]: SOCKET_TX_DONE, 195 bytes sent
[DBG ][ns_wrap]: SOCKET_DATA, sock=5, bytes=112
[DBG ][ns_wrap]: SOCKET_DATA, sock=5, bytes=384
[DBG ][ns_wrap]: SOCKET_DATA, sock=5, bytes=202
[DBG ][ns_wrap]: SOCKET_DATA, sock=5, bytes=178
[DBG ][ns_wrap]: SOCKET_DATA, sock=5, bytes=25
[DBG ][ns_wrap]: SOCKET_TX_DONE, 195 bytes sent
[DBG ][ns_wrap]: SOCKET_DATA, sock=5, bytes=112
[DBG ][ns_wrap]: SOCKET_DATA, sock=5, bytes=384
[DBG ][ns_wrap]: SOCKET_DATA, sock=5, bytes=202
[DBG ][ns_wrap]: SOCKET_DATA, sock=5, bytes=178
[DBG ][ns_wrap]: SOCKET_DATA, sock=5, bytes=25
[DBG ][ns_wrap]: SOCKET_TX_DONE, 195 bytes sent
[DBG ][ns_wrap]: SOCKET_DATA, sock=5, bytes=112
[DBG ][ns_wrap]: SOCKET_DATA, sock=5, bytes=384
[DBG ][ns_wrap]: SOCKET_DATA, sock=5, bytes=202
[DBG ][ns_wrap]: SOCKET_DATA, sock=5, bytes=178
[DBG ][ns_wrap]: SOCKET_DATA, sock=5, bytes=25
[DBG ][ns_wrap]: SOCKET_TX_DONE, 195 bytes sent
[DBG ][ns_wrap]: SOCKET_DATA, sock=5, bytes=112
[DBG ][ns_wrap]: SOCKET_DATA, sock=5, bytes=384
[DBG ][ns_wrap]: SOCKET_DATA, sock=5, bytes=202
[DBG ][ns_wrap]: SOCKET_DATA, sock=5, bytes=178
[DBG ][ns_wrap]: SOCKET_DATA, sock=5, bytes=25
[ERR ][mbedclient]: error() 6

------
mbed-client-example-6lowpan 0.0.2
┣─ mbed-client 1.2.0
┃ ┣─ mbed-client-c 1.1.1 yotta_modules/mbed-client-c
┃ ┃ ┗─ nanostack-libservice 3.0.8 yotta_modules/nanostack-libservice
┃ ┣─ mbed-client-mbed-os 1.1.0 yotta_modules/mbed-client-mbed-os
┃ ┃ ┣─ mbed-drivers 0.11.1 yotta_modules/mbed-drivers
┃ ┃ ┃ ┣─ mbed-hal 1.0.4 yotta_modules/mbed-hal
┃ ┃ ┃ ┃ ┗─ mbed-hal-freescale 1.0.0 yotta_modules/mbed-hal-freescale
┃ ┃ ┃ ┃   ┗─ mbed-hal-ksdk-mcu 1.0.3 yotta_modules/mbed-hal-ksdk-mcu
┃ ┃ ┃ ┃     ┣─ uvisor-lib 1.0.6 yotta_modules/uvisor-lib
┃ ┃ ┃ ┃     ┗─ mbed-hal-k64f 1.0.1 yotta_modules/mbed-hal-k64f
┃ ┃ ┃ ┃       ┗─ mbed-hal-frdm-k64f 1.0.0 yotta_modules/mbed-hal-frdm-k64f
┃ ┃ ┃ ┣─ cmsis-core 1.0.1 yotta_modules/cmsis-core
┃ ┃ ┃ ┃ ┗─ cmsis-core-freescale 1.0.0 yotta_modules/cmsis-core-freescale
┃ ┃ ┃ ┃   ┗─ cmsis-core-k64f 1.0.0 yotta_modules/cmsis-core-k64f
┃ ┃ ┃ ┣─ ualloc 1.0.2 yotta_modules/ualloc
┃ ┃ ┃ ┃ ┗─ dlmalloc 1.0.0 yotta_modules/dlmalloc
┃ ┃ ┃ ┣─ minar 1.0.1 yotta_modules/minar
┃ ┃ ┃ ┃ ┗─ minar-platform 1.0.0 yotta_modules/minar-platform
┃ ┃ ┃ ┃   ┗─ minar-platform-mbed 1.0.0 yotta_modules/minar-platform-mbed
┃ ┃ ┃ ┣─ core-util 1.0.1 yotta_modules/core-util
┃ ┃ ┃ ┗─ compiler-polyfill 1.1.1 yotta_modules/compiler-polyfill
┃ ┃ ┗─ sockets 1.0.2 yotta_modules/sockets
┃ ┃   ┗─ sal 1.0.2 yotta_modules/sal
┃ ┗─ mbed-client-mbedtls 1.0.7 yotta_modules/mbed-client-mbedtls
┃   ┗─ mbedtls 2.2.0-rc.2 yotta_modules/mbedtls
┣─ atmel-rf-driver 1.0.1
┃ ┗─ sal-stack-nanostack 3.0.5 yotta_modules/sal-stack-nanostack
┃   ┣─ sal-stack-nanostack-eventloop 1.0.3 yotta_modules/sal-stack-nanostack-eventloop
┃   ┃ ┗─ mbed-6lowpan-eventloop-adaptor 1.0.2 yotta_modules/mbed-6lowpan-eventloop-adaptor
┃   ┗─ nanostack-randlib 1.0.0 yotta_modules/nanostack-randlib
┗─ mbed-mesh-api 1.0.1
  ┗─ sal-iface-6lowpan 1.0.6 yotta_modules/sal-iface-6lowpan
~~~~

rainierwolfcastle, 2015-11-02:
ARM Internal Ref: ONME-2077

SeppoTakalo, 2015-11-03:
Fixed.

closed 2015-11-03 by SeppoTakalo