ARMmbed/mbed-client-example-6lowpan #41: Code duplication in mbed-client-example-6lowpan, mbed-client-linux-example & mbed-client-linux-examples

markus-becker-tridonic-com, 2015-11-12:
The MbedClient class is duplicated in mbed-client-example-6lowpan, mbed-client-linux-example & mbed-client-examples, it would make sense to factor it out into mbed-client-common module and pull that in from the 3 mentioned modules.

Furthermore, it would make sense to only have the network specific parts in the modules, and the remainder of the setup in the common module.

rainierwolfcastle, 2015-11-13:
ARM Internal Ref: ONME-2143

artokin, 2015-11-19:
Examples are very likely to be updated in the future and separate applications might not be needed anymore. Therefore we are not planning to optimize structure of existing example applications.

closed 2015-11-19 by artokin

markus-becker-tridonic-com, 2015-11-20:
"In the future" == days, weeks, months?

artokin, 2015-11-20:
months :-)

markus-becker-tridonic-com, 2015-11-20:
If we provide a PR within days, would that be accepted? Please talk to @devran01 .

yogpan01, 2015-11-20:
Hi, This is a simple standalone example which contains all information which is needed by this example to work. The common component is all these in mbed-client module, all other information is application logic which is easy to change if they are part of this package and not somewhere else. 
What additional benefit do we get to create separate module only for examples which is just a starting place to get developers started ? 
Anyway, if you provide PR, we can take a look into that and possibly take it in , if it works across all examples.