ARMmbed/mbed-client-example-6lowpan #45: non conform ipv6 representation

thcu-gp, 2015-12-10:
in file source/mbedclient.cpp the ipv6 address for the embed server is hard coded. The format is incorrect and confusing due to scheme/port mixing and the port/ipv6 colon mixing
- The ipv6 address should be enclosed in square brackets
- The scheme does not match the default coap/coaps ports
     coap should use 5683
     coaps should use 5684

current representation: 
coap://2607:f0d0:2601:52::20:5684

possible corrections
coaps://[2607:f0d0:2601:52::20]:5684
coap://[2607:f0d0:2601:52::20]:5683

anttiylitokola, 2015-12-20:
Thanks for your report. We will look into this.

mlnx, 2016-01-25:
cc @mlnx

kmdewaal, 2016-02-05:
This is documented in RFC3986 paragraph 3.2.2

>  A host identified by an Internet Protocol literal address, version 6
>    [RFC3513] or later, is distinguished by enclosing the IP literal
>    within square brackets ("[" and "]").  This is the only place where
>    square bracket characters are allowed in the URI syntax.

mlnx, 2016-03-11:
This has been been fixed on this commit: 56b0a6586046ce057acef2ba9f8574973facc9cd

closed 2016-03-11 by mlnx