ARMmbed/mbed-client-example-6lowpan #46: CoAP Response Code 4.00 From connector.mbed.com

z-mayer, 2015-12-10:
I’m trying to run the example using the following boards:
- FRDM-K64F
- mbed 6LoWPAN Gateway: HPI-0333A
- mbed 6LoWPAN Shield: HPI-0332A

I'm using the security credentials obtained from https://connector.mbed.com/#credentials.

The registration is initiated by the client and the DTLS handshake is successful. 
After sending the first packet of encrypted application data, the following message is printed: error 3.

The device doesn’t show up under Connected Devices on the website.
In a debug session, I can see that the device receives a CoAP Response Code 4.00 (Bad Request). 
The experiment gives the same result with 6LoWPAN ND and Thread type of bootstrap. 
_
[client_uart_output.txt](https://github.com/ARMmbed/mbed-client-example-6lowpan/files/57962/client_uart_output.txt)

The following is a Wireshark capture file, but Github won't let me upload a .zip today. 
Please change the extension to pcapng.
[change_extension_to_pcapng.txt](https://github.com/ARMmbed/mbed-client-example-6lowpan/files/57963/change_extension_to_pcapng.txt)

z-mayer, 2015-12-14:
I was able to get the example working, I was trying to register at the wrong IPv6 address earlier.

While following the 'Client side configuration' section in the README, I changed the MBED_DEVICE_CONNECTOR_URI in mbedclient.cpp to "coap://2607:f0d0:3701:9f::20:5684". 
I did not suspect that the address was wrong, because the DTLS handshake was successful.

Please update the README with the correct address.

artokin, 2015-12-14:
Good that the problem was found!
The README has now been updated to use correct IPv6 address "2607:f0d0:2601:52::20". Sorry that README was not accurate.

SeppoTakalo, 2015-12-14:
Good finding. I think that we have accidentally left our test server address in the README. It really can do the same handshake but will drop the connection because certification does not match.

closed 2015-12-14 by SeppoTakalo