ARMmbed/mbed-client-example-6lowpan #48: Scan Fail: No Beacons

cyliangtw, 2016-01-28:
I followed Troubleshooting steps, still can't identify the failure.
(1) Set as 6LoWPAN ND of mesh network, I got "Scan Fail: No Beacons" repeatedly.
(2) Set as Thread type of mesh network, I got "ns_sal_socket_send_to: error=-4" repeatedly.

Details of log as below:
=====Log (6LoWPAN ND of mesh network)  =====================
[DBG ][m6La]: init()
[DBG ][m6La]: connect()
[INFO][m6LND]: Start 6LoWPAN ND Bootstrap
[DBG ][m6LND]: app_parse_network_event() 3
[DBG ][m6LND]: Link Layer Scan Fail: No Beacons
[DBG ][m6LND]: Restart bootstrap
[DBG ][m6LND]: app_parse_network_event() 3
[DBG ][m6LND]: Link Layer Scan Fail: No Beacons

===== Log (Thread type of mesh network)  =====================
Start mbed-client-example-6lowpan
[DBG ][m6La]: init()
[DBG ][m6La]: connect()
[INFO][m6Thread]: Start 6LoWPAN Thread bootstrap
[DBG ][m6Thread]: app_parse_network_event() 0
[INFO][m6Thread]: Thread bootstrap ready
[DBG ][m6Thread]: MAC 16-bit: 00:00
[DBG ][m6Thread]: PAN ID: fa:ce
[DBG ][m6Thread]: MAC 64-bit: fe:c2:3d:00:00:05:20:70
[DBG ][m6Thread]: IID (Based on MAC 64-bit address): fc:c2:3d:00:00:05:20:70
                                                                            mesh
_network_handler() 0
[DBG ][ns_wrap]: ns_wrapper_socket_open(7)
                                          waiting 5s before sending registration
...
send_registration()
[ERR ][ns_sal]: ns_sal_socket_send_to: error=-4
                                               error 6
Reconnecting to server
[DBG ][ns_wrap]: ns_wrapper_socket_free(7)
[DBG ][ns_wrap]: ns_wrapper_socket_open(8)
                                          waiting 5s before sending registration
...
send_registration()
[ERR ][ns_sal]: ns_sal_socket_send_to: error=-4
                                               error 6

cyliangtw, 2016-01-29:
I powered up the border router "FireFly mbed 6LoWPAN Gateway router' by  micro-USB cable.
I also copied  gateway binary firmware file to flash this gateway device successfully by  micro-USB cable power up.

SeppoTakalo, 2016-01-29:
Hi,

Two things I would like to know:
* Which binary file did you use to flash the device?
* Is the border router attached to working ethernet segment?

| Binary  | Use case |
| --------- | ------------- |
| gateway6LoWPANDynamic.bin  | 6LoWPAN border router in bridge mode. Learns IPv6 prefix from Ethernet  |
| gateway6LoWPANStatic.bin |  6LoWPAN border router using static IPv6 prefix |
| gatewayThreadDynamic.bin | Thread border router in bridge mode. Learns IPv6 prefix from Ethernet |
| gatewayThreadStatic.bin | Thread border router using static IPv6 prefix |

What is common to all those routers is it they require ethernet cable to be attached to switch, or another computer before they boot up.

Those routers that learns the prefix from the Ethernet side, requires that there is active IPv6 gateway in the ethernet segment and it is sending ICMPv6 Router advertisements with the A flag (autonomous address-configuration flag) turned on.

In case you do not have IPv6 routing in your network, try out first with the static ones.

cyliangtw, 2016-01-29:
Hi,
1. I tried gateway binary files: both of gateway6LoWPANDynamic and gatewayThreadDynamic.
2. The border router attached one working ethernet segment. I verified OK by "mbed-client-examples-master" sample.
After applied gateway6LoWPANStatic, it could continuously run until send_registration() & got error 6.
I applied the same security.h of "mbed-client-examples-master" except keeping PSK & PSK_IDENTITY.
Is there anything missing ?
-----------------------------------------------------------------------------
[DBG ][m6LND]: PAN ID: 06:91
[DBG ][m6LND]: MAC 64-bit: fc:c2:3d:00:00:05:20:70
[DBG ][m6LND]: IID (Based on MAC 64-bit address): fe:c2:3d:00:00:05:20:70
[DBG ][m6LND]: Channel: 12
                          mesh_network_handler() 0
[DBG ][ns_wrap]: ns_wrapper_socket_open(1)
                                          waiting 5s before sending registration
...
send_registration()
[DBG ][ns_wrap]: SOCKET_TX_DONE, 163 bytes sent
[DBG ][ns_wrap]: SOCKET_TX_DONE, 163 bytes sent
[DBG ][ns_wrap]: SOCKET_TX_DONE, 163 bytes sent
                                               error 6
Reconnecting to server
[DBG ][ns_wrap]: ns_wrapper_socket_free(1)
[DBG ][ns_wrap]: ns_wrapper_socket_open(2)

SeppoTakalo, 2016-01-29:
That explains it.
You have nothing wrong in your code or boards.

The *dynamic* versions require **IPv6** connectivity which you most probably do not have. The *mbed_client_example* you used on K64F over ethernet, uses **IPv4**

The *static* versions of the router are provided only for testing purposes. They form site-local *IPv6* network that cannot be routed outside and therefore your registration fails after "send_registration()" when it receives ICMP error message from router that it cannot route. 

Maybe the documentation does not state it clear enough that *6LoWPAN* and *Thread* require **IPv6**

[image]

* https://github.com/ARMmbed/mbed-client-example-6lowpan#ip-address-setup

closed 2016-01-29 by SeppoTakalo

cyliangtw, 2016-01-30:
Thanks a lot, I realized thread rely on IPv6. In network topology of IoT,  IoT  gateway role will be in charge of the convert between IPv6 and IPv4, after that IoT host device could connect nomal internet through IoT borader router.
    If I use Google OnHub instead of FireFly gateway, could it work to connect ARM connector ?

- ������Sony Xperia����������������

---- Seppo Takalo���� ----

That explains it.
You have nothing wrong in your code or boards.

The dynamic versions require IPv6 connectivity which you most probably do not have. The mbed_client_example you used on K64F over ethernet, uses IPv4

The static versions of the router are provided only for testing purposes. They form site-local IPv6 network that cannot be routed outside and therefore your registration fails after "send_registration()" when it receives ICMP error message from router that it cannot route.

Maybe the documentation does not state it clear enough that 6LoWPAN and Thread require IPv6

[network setup]<https://github.com/ARMmbed/mbed-client-example-6lowpan/raw/master/img/Dia2.png>

  *   https://github.com/ARMmbed/mbed-client-example-6lowpan#ip-address-setup

-

===========================================================================================
The privileged confidential information contained in this email is intended for use only by the addressees as indicated by the original sender of this email. If you are not the addressee indicated in this email or are not responsible for delivery of the email to such a person, please kindly reply to the sender indicating this fact and delete all copies of it from your computer and network server immediately. Your cooperation is highly appreciated. It is advised that any unauthorized use of confidential information of Nuvoton is strictly prohibited; and any information in this email irrelevant to the official business of Nuvoton shall be deemed as neither given nor endorsed by Nuvoton.