ARMmbed/mbed-client-example-6lowpan #50: linking in sal-stack-nanostack-private results in build issues

thcu-gp, 2016-01-29:
@mlnx 

linking in sal-stack-nanostack-private results in build issues:
(and the list of warnings is large)

error: mbed-client-c does not meet specification ^1.1.0 required by coap-service
error: mbed-drivers does not meet specification ~0.11.1 required by core-util
...
C:/mbed/vlatest/mbed-client-example-6lowpan/yotta_modules/atmel-rf-driver/source/driverRFPhy.c: In function 'rf_device_register':
C:/mbed/vlatest/mbed-client-example-6lowpan/yotta_modules/atmel-rf-driver/source/driverRFPhy.c:123:25: error: request for member 'channel_count' in something not a structure or union
             channel_info.channel_count = 11;
                         ^
C:/mbed/vlatest/mbed-client-example-6lowpan/yotta_modules/atmel-rf-driver/source/driverRFPhy.c:125:25: error: request for member 'channel_mask' in something not a structure or union
             channel_info.channel_mask = 0x000007ff;
                         ^
C:/mbed/vlatest/mbed-client-example-6lowpan/yotta_modules/atmel-rf-driver/source/driverRFPhy.c:132:25: error: request for member 'channel_count' in something not a structure or union
             channel_info.channel_count = 16;
                         ^
C:/mbed/vlatest/mbed-client-example-6lowpan/yotta_modules/atmel-rf-driver/source/driverRFPhy.c:134:25: error: request for member 'channel_mask' in something not a structure or union
             channel_info.channel_mask = 0x07FFF800;
                         ^
C:/mbed/vlatest/mbed-client-example-6lowpan/yotta_modules/atmel-rf-driver/source/driverRFPhy.c:138:22: error: 'phy_device_driver_s' has no member named 'link_channel_info'
         device_driver.link_channel_info = &channel_info;
                      ^

manually linking in atmel drivers from a github repo then results in:
C:/mbed/vlatest/mbed-client-example-6lowpan/yotta_modules/mbed-mesh-api/source/thread_tasklet.c: In function 'thread_tasklet_configure_and_connect_to_network':
C:/mbed/vlatest/mbed-client-example-6lowpan/yotta_modules/mbed-mesh-api/source/thread_tasklet.c:215:41: error: 'link_configuration_s' has no member named 'steering_data_len'
     thread_tasklet_data_ptr->link_config.steering_data_len = 0;
                                         ^
C:/mbed/vlatest/mbed-client-example-6lowpan/yotta_modules/mbed-mesh-api/source/thread_tasklet.c:233:5: warning: implicit declaration of function 'thread_managenet_node_init' [-Wimplicit-f
unction-declaration]
     thread_managenet_node_init(thread_tasklet_data_ptr->nwk_if_id,
     ^
C:/mbed/vlatest/mbed-client-example-6lowpan/yotta_modules/mbed-mesh-api/source/thread_tasklet.c: In function 'thread_tasklet_set_device_config':
C:/mbed/vlatest/mbed-client-example-6lowpan/yotta_modules/mbed-mesh-api/source/thread_tasklet.c:360:12: warning: pointer targets in passing argument 1 of 'strcpy' differ in signedness [-W
pointer-sign]
     strcpy(device_configuration.PSKd_ptr, pskd);
            ^
In file included from c:\progra~2\gnutoo~1\4bd65~1.920\arm-none-eabi\include\string.h:10:0,
                 from C:/mbed/vlatest/mbed-client-example-6lowpan/yotta_modules/mbed-mesh-api/source/thread_tasklet.c:17:
c:\progra~2\gnutoo~1\4bd65~1.920\arm-none-eabi\include\string.h:30:8: note: expected 'char * restrict' but argument is of type 'uint8_t *'
 char  *_EXFUN(strcpy,(char *__restrict, const char *__restrict));
        ^
C:/mbed/vlatest/mbed-client-example-6lowpan/yotta_modules/mbed-mesh-api/source/thread_tasklet.c:362:25: error: 'device_configuration_s' has no member named 'leaderCap'
     device_configuration.leaderCap = true;
                         ^

SeppoTakalo, 2016-01-29:
If you link against our development repository, please checkout the version tag that has been released for Yotta.
In this case, it would be `git checkout v3.0.5`

The development repository is always a work in progress and we cannot guarantee that *master* branch builds with other modules.

mlnx, 2016-01-29:
Indeed,
@thcu-gp you may want to look at some scripts to clone and link repositories, for example:
https://github.com/ARMmbed/sal-stack-nanostack/files/108212/setup_nanostack_source.txt

mlnx, 2016-02-04:
@thcu-gp have you managed to fix the problem?

thcu-gp, 2016-02-08:
by fixing the tag of the sal-nanostack to 3.0.5, I am able to build it. A lot of warnings keep popping up though (mostly unused values and unused parameter warnings).

mlnx, 2016-02-08:
Just tested the example with nanostack v3.0.5 and does not fail (a binary is generated)

thcu-gp, 2016-02-08:
apparantly some stuff did not get updated while swapping nanostrack-private versions.
on last try, I could chechout the listed versions on github
I had to manully link in higher versions of sal-stack-nanostack-eventloop and mbed-6lowpan-eventloop-adaptor to get nanostack v3.0.5 working (but it works now)

closed 2016-02-10 by mlnx