ARMmbed/mbed-client-example-6lowpan #65: Could I verify without gateway/ border router ?

SYuPing, 2016-03-14:
I have same problem with #48 , also I have no gateway router yet...
Actully, I've order An mbed 6LoWPAN gateway router which is required, but still on the plane...
So, I have several quention want to know...

Btw, I'm using the RF module which named uz2400 also follow the IEEE 802.15.4 standard instead of atmel RF module as reserve, and also updated  source codes inside atmel-rf-driver.
 
(1) Whether this example couldn't run only client end-point without gateway router or not? 
     Because I want to verify the RF module could work well (including does it could register to network?, work well by using thread mesh network...) before the gateway arrived.

(2) Is there any suggestions or examples for help me to verify the RF module on this thread example could work well step by step?

(3) I've also tried to use k64f be a border router, but I still couldn't find the web page, 
https://github.com/ARMmbed/k64f-border-router, did it closed?

SeppoTakalo, 2016-03-14:
Hi @SYuPing 

Thank you for your report. Please find my answer to your questions.

(1) This example requires border router to run as there is direct communication with the mbed Connector service.

(2) Verifying that your RF driver work might require you to have RF sniffer. With the border router, you can ping from the external network, if you know the IPv6 address that the end node will use. For definitely verifying the stability and correct behaviour of the driver, we do not have any publicly available methods.

(3) The K64F border router will be released with the upcoming mbed OS 16 Q1 release.
We are currently preparing the release and therefore the repositories are open only for mbed OS partners. I'm sorry for the inconvenience that we might have caused by prematurely publishing links to non-open repositories.

I will close this issue for now, as we use the issue list to track software related bugs. But please feel free to ask more questions by replying this thread.

Best regards,
Seppo Takalo.

closed 2016-03-14 by SeppoTakalo

SYuPing, 2016-03-15:
OK!
Thanks a lot!