ARMmbed/mbed-client-example-6lowpan #66: Examples for arm_random_module_init(void) / arm_random_seed_get(void)

markus-becker-tridonic-com, 2016-03-14:
Is there documentation or sample code for at least one platform available for the functions  arm_random_module_init(void) and arm_random_seed_get(void)? Is it correct that, rand() would not produce anything random at all, if the target platform has no RTC?

markus-becker-tridonic-com, 2016-04-04:
@mlnx would you follow up on this?

kjbracey-arm, 2016-04-04:
Typically arm_random_module_init() is a no-op, but it can be used to make sure whatever hardware you need to get a random seed is active.
 
arm_random_seed_get() should return a best effort real random number seed, that will vary from board to board, and from boot to boot on a given board. Some sort of combination of the MAC address, and some radio noise read from the radio driver would be appropriate, and is what we normally use. RTC if available would be another possibility.

This call is only made once at start-up, to seed srand(). Speed is not critical.
 
Many of our RF drivers provide a rf_read_random() call which gives 8 bits of randomness, but note that the radio hardware must be brought up in time to ensure this works.

markus-becker-tridonic-com, 2016-04-06:
Is there a sample for a specific target? The API of it not documented.

SeppoTakalo, 2016-04-06:
I don't believe that there is any samples for mbed OS.
Eventually we will migrate into using random modules from mbed TLS, so these are just legacy porting layer between Nanostack and the platform it is running on.

markus-becker-tridonic-com, 2016-04-07:
When will this change? Does it still make sense to implement it?

TuomoHautamaki, 2016-04-14:
This change is not yet scheduled, so it is not coming in a near future. We recommend to proceed without (us using random modules from mbed TLS)

markus-becker-tridonic-com, 2016-04-22:
If this is not scheduled yet, is it possible to show sample code for one target, that once can use for inspiration,implementation and usage, while the mbed-tls solution is being worked on?

kjbracey-arm, 2016-04-22:
Here's a (not-very-good) implementation that will work with the Atmel RF driver (or any other that provided the same rf_ calls)

    uint32_t arm_random_seed_get(void)
    {
        uint8_t tmp_mac_addr[8];
        uint32_t rndtmp = rf_read_random();
        uint8_t i;

        rf_read_mac_address(tmp_mac_addr);
        for (i = 0; i < 8; i++) {
            rndtmp += tmp_mac_addr[i];
        }

        return rndtmp;
    }

Main potential pitfall is on start-up order - rf_read_random() is reading a value that was stored during rf_init(), so if this gets called before rf_init(), there will be no radio randomness.

closed 2016-05-11 by SeppoTakalo