ARMmbed/mbed-client-example-6lowpan #69: mbed-client-example-6lowpan : yotta reports dependencies are not met

shahab5, 2016-04-15:
`yotta target frdm-k64f-gcc`
`yotta install mbed-client-example-6lowpan`

info: get versions for frdm-k64f-gcc
info: download frdm-k64f-gcc@2.0.0 from the public module registry
info: get versions for kinetis-k64-gcc
info: download kinetis-k64-gcc@2.0.2 from the public module registry
info: get versions for mbed-gcc
info: download mbed-gcc@1.2.2 from the public module registry
[root@cpu335 mbed-client-example-6lowpan]# 
[root@cpu335 mbed-client-example-6lowpan]# yt build
info: get versions for mbed-client-mbed-os
info: download mbed-client-mbed-os@0.1.11 from the public module registry
info: get versions for mbed-drivers
info: download mbed-drivers@0.6.9 from the public module registry
info: get versions for sockets
info: download sockets@0.3.6 from the public module registry
info: get versions for mbed-hal
info: download mbed-hal@0.6.4 from the public module registry
info: get versions for cmsis-core
info: download cmsis-core@0.2.7 from the public module registry
info: get versions for ualloc
info: download ualloc@0.0.10 from the public module registry
info: get versions for minar
info: download minar@0.6.7 from the public module registry
info: get versions for core-util
info: download core-util@0.0.16 from the public module registry
info: get versions for mbed-hal-freescale
info: download mbed-hal-freescale@0.5.2 from the public module registry
info: get versions for mbed-hal-ksdk-mcu
info: download mbed-hal-ksdk-mcu@0.5.7 from the public module registry
info: get versions for uvisor-lib
info: download uvisor-lib@0.7.25 from the public module registry
info: get versions for mbed-hal-k64f
info: download mbed-hal-k64f@0.3.6 from the public module registry
info: get versions for mbed-hal-frdm-k64f
info: download mbed-hal-frdm-k64f@0.4.6 from the public module registry
info: get versions for cmsis-core-freescale
info: download cmsis-core-freescale@0.1.4 from the public module registry
info: get versions for cmsis-core-k64f
info: download cmsis-core-k64f@0.1.5 from the public module registry
info: get versions for dlmalloc
info: download dlmalloc@0.0.6 from the public module registry
info: get versions for compiler-polyfill
info: download compiler-polyfill@1.0.4 from the public module registry
info: get versions for minar-platform
info: download minar-platform@0.3.4 from the public module registry
info: get versions for minar-platform-mbed
info: download minar-platform-mbed@0.1.5 from the public module registry
info: get versions for sal
info: download sal@0.2.3 from the public module registry
info: get versions for sal-stack-lwip
info: download sal-stack-lwip@0.2.8 from the public module registry
info: get versions for sal-driver-lwip-k64f-eth
info: download sal-driver-lwip-k64f-eth@0.1.3 from the public module registry
info: get versions for sal-iface-eth
info: download sal-iface-eth@0.1.4 from the public module registry

**error: mbed-client-c does not meet specification ^2.0.0 required by coap-service
error: mbed-drivers does not meet specification ^1.0.0 required by mbed-6lowpan-eventloop-adaptor
error: minar does not meet specification ^1.0.0 required by sal-stack-nanostack-eventloop**

teetak01, 2016-04-15:
Hi, most of the modules you are downloading seems to be very old. Do you happen to have some old release of mbed-client-example-6lowpan? You probably should get the newest version from github:

Yotta install is not required.

git clone git@github.com:ARMmbed/mbed-client-example-6lowpan.git

Few things to verify that you have newest build:

```
git pull (git stash && git stash pop to keep the security.h file out of way)
yt clean
rm -rf yotta_modules 
rm -rf yotta_targets   (might be required to get the newest target)
yt target frdm-k64f-gcc
yt build
```

shahab5, 2016-04-15:
Thanks! It worked.

Used:
`git clone git@github.com:ARMmbed/mbed-client-example-6lowpan.git`

closed 2016-04-15 by shahab5

teetak01, 2016-04-15:
Good!

Very true, this would not require public key:

`git clone https://github.com/ARMmbed/mbed-client-linux-example.git`