ARMmbed/mbed-client-example-6lowpan #71: mbed-client unable to connect to GW router

shahab5, 2016-04-21:
Using FRDM-K64 board as client (with Firefly RF shield) and Firefly GW router.

**GW_ROUTER debug log**

[DBG ][core]: Allocate Root Tasklet
P.Init
[DBG ][pana]: Init
[DBG ][sck ]: Socket id 0 allocated
Starting NanoRouter in autonomous mode, extension=1
[DBG ][core]: NS Root task Init
[DBG ][sck ]: Socket Tasklet Generated
[DBG ][mle ]: MLE Tasklet Generated
[DBG ][pana]: Pana Tasklet Init
[DBG ][board]: Setting up TX power to be 0x0F+Amp = +4.1 dBm
[INFO][EP  ]: endpoint_tasklet(): init
[DBG ][sck ]: Socket id 1 allocated
[DBG ][sck ]: Socket id 2 allocated
[DBG ][ip6s]: IPV6 interface Base Ready
[DBG ][rout]: Added route:
[DBG ][rout]:                    fe80::/64  if:1 src:'Static' id:0 lifetime:infinite
[DBG ][rout]:      On-link (met 128)
Ipv6 bootsrap started
[INFO][addr]: Tentative Address added to IF 1: fe80::fec2:3dff:fe05:347f
[DBG ][addr]: LL64 Register OK!
[INFO][addr]: DAD passed on IF 1: fe80::fec2:3dff:fe05:347f
[DBG ][ip6s]: Interface ID: 1, ipv6: fe80::fec2:3dff:fe05:347f
[DBG ][ip6s]: DAD OK
[DBG ][ip6s]: Push RS to IPV6
[DBG ][ip6s]: Push RS to IPV6
[DBG ][ip6s]: Push RS to IPV6
[DBG ][ip6s]: Push RS to IPV6
Destination Cache:
Routing table:
                   fe80::/64  if:1 src:'Static' id:0 lifetime:infinite
     On-link (met 128)
RPL memory usage 36
RPL Domain 20007bb4
Neighbour Cache 1
Reachable Time: 38733   Retrans Timer: 1000   MTU: 1500
Neighbour Cache 0
Reachable Time: 26351   Retrans Timer: 1000   MTU: 1280
[DBG ][ip6s]: Push RS to IPV6
[DBG ][ip6s]: Push RS to IPV6
Destination Cache:
Routing table:
                   fe80::/64  if:1 src:'Static' id:0 lifetime:infinite
     On-link (met 128)
RPL memory usage 36
RPL Domain 20007bb4
Neighbour Cache 1
Reachable Time: 38733   Retrans Timer: 1000   MTU: 1500
Neighbour Cache 0
Reachable Time: 26351   Retrans Timer: 1000   MTU: 1280
[DBG ][ip6s]: Push RS to IPV6
[DBG ][ip6s]: Push RS to IPV6
Destination Cache:
Routing table:
                   fe80::/64  if:1 src:'Static' id:0 lifetime:infinite
     On-link (met 128)

shahab5, 2016-04-21:
**Client Debug Trace** 
Start mbed-client-example-6lowpan
[DBG ][m6La]: init()
[DBG ][m6La]: connect()
[DBG ][m6LND]: Link-layer security NOT enabled.
[DBG ][m6LND]: Channel: 12
[DBG ][m6LND]: Channel page: 0
[DBG ][m6LND]: Channel mask: 2048
[INFO][m6LND]: Start 6LoWPAN ND Bootstrap
[DBG ][m6LND]: app_parse_network_event() 3
[DBG ][m6LND]: Link Layer Scan Fail: No Beacons
[DBG ][m6LND]: Restart bootstrap
[DBG ][m6LND]: app_parse_network_event() 3
[DBG ][m6LND]: Link Layer Scan Fail: No Beacons

shahab5, 2016-04-21:
Using `gateway6LoWPANDynamic.bin ` binary for GW router.

**Question: **
How to configure GW router radio channel? 
Does config.json configure Radio channel for both mbed client and GW router?

shahab5, 2016-04-21:
Sorry, I missed seeing https://github.com/ARMmbed/mbed-client-example-6lowpan/issues/48

Tried gateway6LoWPANStatic.bin its working, so its got to be ipv6 issue.

Kindly, close this by mentioning, way to configure GW_router radio settings.

Thanks

artokin, 2016-04-25:
Hi Shahab,

The mbed 6LoWPAN gateway is statically configured to operate on channel 12 (2.4GHz RF) or on channel 1 (sub-GHz RF). It is not possible to configure channels in the mbed 6LoWPAN GW.

Another 6LoWPAN GW is available in https://github.com/ARMmbed/k64f-border-router that allows you to use FRDM-K64 as a border router. It allows channel configuiration.

BR,
Arto

shahab5, 2016-04-25:
Hi Arto,

Thank a lot for your reply.
Closing this issue.

Regards,
Shahab

closed 2016-04-25 by shahab5