ARMmbed/mbed-client-example-6lowpan #74: No entries in mbed device connector

gdillen, 2016-05-03:
Hello,

I followed the instructions for setting up the connection between my FRDM-K64F with Firefly shield 6lowpan edge device and my Firefly 6lowpan gateway. My ISP supports IPv6 and I have a router with switch that also support IPv6. Both edge device and gateway are connected via USB to my laptop, the gateway is Ethernet-connected to my switch.
I tried with both Dynamic and Static firmware on my gateway. No data is entering mbed device connector. See Putty logs below.

Anyone an idea what is going wrong here?

Many thanks;
Guy

**Putty log coming from gateway (6lowpan static firmware)
**
[DBG ][6Bo ]: RS*
[DBG ][rplc]: transmit dio, rank: 80
[DBG ][6Bo ]: RS*
[DBG ][rplc]: transmit dio, rank: 80
[DBG ][br  ]: Border router Backhaul link ready
[DBG ][prox]: Proxy bridge enabled for interface 1 to 0

6Lowpan bootstrap ready, IP=fd00:ff1:ce0b:a5e0:0:ff:fe00:57d5
Destination Cache:
Routing table:
                   fe80::/64  if:0 src:'Static' id:0 lifetime:infinite
     On-link (met 128)
 fd00:ff1:ce0b:a5e0:0:ff:fe00:57d5/128 if:0 src:'Loopback' id:0 lifetime:infinite
     On-link (met 128)
 fd00:ff1:ce0b:a5e0:fec2:3dff:fe04:803a/128 if:1 src:'Loopback' id:0 lifetime:infinite
     On-link (met 128)
                   fe80::/64  if:1 src:'Static' id:0 lifetime:infinite
     On-link (met 128)
     fd00:ff1:ce0b:a5e0::/64  if:1 src:'Static' id:0 lifetime:1785
     On-link (met 64)
RPL memory usage 336
RPL Domain 20007bc0
RPL Instance 1
---------------
DODAG fd00:ff1:ce0b:a5e0:0:ff:fe00:57d5
  G=0 MOP=1 Prf=7
    fd00:ff1:ce0b:a5e0::/64  lifetime:infinite pref:-1
                      ::/0   lifetime:infinite pref:0
    fd00:ff1:ce0b:a5e0::/64  lifetime:infinite flags:--*
  Version 240
  *Current version* Rank=0080
DIO trickle Imin=6, Imax=20972, k=10
            I=96, now=56, t=63, c=0
Neighbour Cache 1
Reachable Time: 32964   Retrans Timer: 1000   MTU: 1500
Neighbour Cache 0
Reachable Time: 27617   Retrans Timer: 1000   MTU: 1280
[DBG ][rplc]: transmit dio, rank: 80
[DBG ][6Bo ]: Send MLE Advertisement
Destination Cache:
Routing table:
                   fe80::/64  if:0 src:'Static' id:0 lifetime:infinite
     On-link (met 128)
 fd00:ff1:ce0b:a5e0:0:ff:fe00:57d5/128 if:0 src:'Loopback' id:0 lifetime:infinite
     On-link (met 128)
 fd00:ff1:ce0b:a5e0:fec2:3dff:fe04:803a/128 if:1 src:'Loopback' id:0 lifetime:infinite
     On-link (met 128)
                   fe80::/64  if:1 src:'Static' id:0 lifetime:infinite
     On-link (met 128)
     fd00:ff1:ce0b:a5e0::/64  if:1 src:'Static' id:0 lifetime:1797
     On-link (met 64)
RPL memory usage 336
RPL Domain 20007bc0

**Putty log from FRDM-K64F edge device**
[DBG ][m6LND]: app_parse_network_event() 8
[DBG ][m6LND]: ND/RPL scan new network
[DBG ][m6LND]: Restart bootstrap
[DBG ][m6LND]: app_parse_network_event() 8
[DBG ][m6LND]: ND/RPL scan new network
[DBG ][m6LND]: Restart bootstrap
[DBG ][m6LND]: app_parse_network_event() 8
[DBG ][m6LND]: ND/RPL scan new network
[DBG ][m6LND]: Restart bootstrap
[DBG ][m6LND]: app_parse_network_event() 8
[DBG ][m6LND]: ND/RPL scan new network
[DBG ][m6LND]: Restart bootstrap
[DBG ][m6LND]: app_parse_network_event() 8
[DBG ][m6LND]: ND/RPL scan new network
[DBG ][m6LND]: Restart bootstrap
[DBG ][m6LND]: app_parse_network_event() 8
[DBG ][m6LND]: ND/RPL scan new network
[DBG ][m6LND]: Restart bootstrap
[DBG ][m6LND]: app_parse_network_event() 8
[DBG ][m6LND]: ND/RPL scan new network
[DBG ][m6LND]: Restart bootstrap
[DBG ][m6LND]: app_parse_network_event() 8
[DBG ][m6LND]: ND/RPL scan new network
[DBG ][m6LND]: Restart bootstrap
[DBG ][m6LND]: app_parse_network_event() 3
[DBG ][m6LND]: Link Layer Scan Fail: No Beacons
[DBG ][m6LND]: Restart bootstrap
[DBG ][m6LND]: app_parse_network_event() 8
[DBG ][m6LND]: ND/RPL scan new network
[DBG ][m6LND]: Restart bootstrap
[DBG ][m6LND]: app_parse_network_event() 8
[DBG ][m6LND]: ND/RPL scan new network
[DBG ][m6LND]: Restart bootstrap
[DBG ][m6LND]: app_parse_network_event() 8
[DBG ][m6LND]: ND/RPL scan new network
[DBG ][m6LND]: Restart bootstrap
[DBG ][m6LND]: app_parse_network_event() 8
[DBG ][m6LND]: ND/RPL scan new network
[DBG ][m6LND]: Restart bootstrap
[DBG ][m6LND]: app_parse_network_event() 8
[DBG ][m6LND]: ND/RPL scan new network
[DBG ][m6LND]: Restart bootstrap
[DBG ][m6LND]: app_parse_network_event() 8
[DBG ][m6LND]: ND/RPL scan new network
[DBG ][m6LND]: Restart bootstrap
[DBG ][m6LND]: app_parse_network_event() 8
[DBG ][m6LND]: ND/RPL scan new network
[DBG ][m6LND]: Restart bootstrap
[DBG ][m6LND]: app_parse_network_event() 8
[DBG ][m6LND]: ND/RPL scan new network
[DBG ][m6LND]: Restart bootstrap

SeppoTakalo, 2016-05-03:
Hi @gdillen 

I see two issues.

First:
Your log files show that you are using the `-static` one for the gateway, therefore it uses statically defined IPv6 prefix `fd00:ff1:ce0b:a5e0` and that is not going to work.
Please use the `-dynamic` one for the GW.

Secondly:
The client device is not joining the 6LoWPAN network.
You can see the output lines: `[DBG ][m6LND]: Link Layer Scan Fail: No Beacons`
Can you make sure that your channel configuration matches between the router and the node.

Br. Seppo

gdillen, 2016-05-03:
Hi @SeppoTakalo 
Thanks.

1. I flashed the "gateway6LoWPANDynamic.bin" firmware to the gateway. Since this is a bin file, does this uses the default channel 12 (that's the one I use on the client)?
2. The channel on the client is the default channel 12

Unfortunately still no success (no data on device connector).

Anything I'm still missing?

Thanks.
Guy

Logfile gateway (I changed part of my IPv6 address by xxxx):

&h▒I▒▒L▒▒▒U▒4I▒_)D▒_i▒k :j$▒w▒N▒BI▒4▒-▒$▒       kpӂ▒▒▒▒\▒+
                                                         ▒▒▒▒t  1"▒04hi*d)j|▒▒\▒`▒▒v'&h)a       ( (▒▒h b(▒▒▒▒;Sx7&.▒▒h▒h&▒▒▒)▒(i▒(^▒$t▒jh"&*▒▒▒▒▒)▒
                                                                   n$▒Destination Cache:
Routing table:
                   fe80::/64  if:0 src:'Static' id:0 lifetime:infinite
     On-link (met 128)
 2a02:a03f:1a30:fd00:0:ff:fe00:4f1/128 if:0 src:'Loopback' id:0 lifetime:infinite
     On-link (met 128)
 2a02:a03f:1a30:fd00:fec2:3dff:fe04:803a/128 if:1 src:'Loopback' id:0 lifetime:infinite
     On-link (met 128)
 fd9f:b3fa:a2dd:0:fec2:3dff:fe04:803a/128 if:1 src:'Loopback' id:0 lifetime:infinite
     On-link (met 128)
    2a02:a03f:1a30:fd00::/56  if:1 src:'RAdv' id:0 lifetime:345563
     next-hop fe80::c256:27ff:fe39:xxxx (met 64)
    2a02:a03f:1a30:fd00::/64  if:1 src:'RAdv' id:0 lifetime:345563
     On-link (met 128)
         fd9f:b3fa:a2dd::/64  if:1 src:'RAdv' id:0 lifetime:7103
     On-link (met 128)
                       ::/0   if:1 src:'RAdv' id:0 lifetime:1703
     next-hop fe80::c256:27ff:fe39:xxxx (met 128)
                   fe80::/64  if:1 src:'Static' id:0 lifetime:infinite
     On-link (met 128)
RPL memory usage 336
RPL Domain 20007bb4
RPL Instance 1
---------------
DODAG 2a02:a03f:1a30:fd00:0:ff:fe00:4f1
  G=0 MOP=1 Prf=7
   2a02:a03f:1a30:fd00::/64  lifetime:infinite pref:-1
                      ::/0   lifetime:infinite pref:0
   2a02:a03f:1a30:fd00::/64  lifetime:infinite flags:--*
  Version 240
  *Current version* Rank=0080
DIO trickle Imin=6, Imax=20972, k=10
            I=768, now=183, t=468, c=0
Neighbour Cache 1
Reachable Time: 36391   Retrans Timer: 1000   MTU: 1280
Router IP Addr: fe80::c256:27ff:fe39:xxxx
LL Addr: (STALE 0) c0:56:27:39:xx:xx
Neighbour Cache 0
Reachable Time: 28999   Retrans Timer: 1000   MTU: 1280
[DBG ][6Bo ]: Send MLE Advertisement
Destination Cache:
Routing table:
                   fe80::/64  if:0 src:'Static' id:0 lifetime:infinite
     On-link (met 128)
 2a02:a03f:1a30:fd00:0:ff:fe00:4f1/128 if:0 src:'Loopback' id:0 lifetime:infinite
     On-link (met 128)
 2a02:a03f:1a30:fd00:fec2:3dff:fe04:803a/128 if:1 src:'Loopback' id:0 lifetime:infinite
     On-link (met 128)
 fd9f:b3fa:a2dd:0:fec2:3dff:fe04:803a/128 if:1 src:'Loopback' id:0 lifetime:infinite
     On-link (met 128)
    2a02:a03f:1a30:fd00::/56  if:1 src:'RAdv' id:0 lifetime:345543
     next-hop fe80::c256:27ff:fe39:xxxx (met 64)
    2a02:a03f:1a30:fd00::/64  if:1 src:'RAdv' id:0 lifetime:345543
     On-link (met 128)
         fd9f:b3fa:a2dd::/64  if:1 src:'RAdv' id:0 lifetime:7083
     On-link (met 128)
                       ::/0   if:1 src:'RAdv' id:0 lifetime:1683
     next-hop fe80::c256:27ff:fe39:xxxx (met 128)
                   fe80::/64  if:1 src:'Static' id:0 lifetime:infinite
     On-link (met 128)
RPL memory usage 336
RPL Domain 20007bb4

Logfile client 

Start mbed-client-example-6lowpan
[DBG ][m6La]: init()
[DBG ][m6La]: connect()
[DBG ][m6LND]: Link-layer security NOT enabled.
[DBG ][m6LND]: Channel: 12
[DBG ][m6LND]: Channel page: 0
[DBG ][m6LND]: Channel mask: 4096
[INFO][m6LND]: Start 6LoWPAN ND Bootstrap
[DBG ][m6LND]: app_parse_network_event() 8
[DBG ][m6LND]: ND/RPL scan new network
[DBG ][m6LND]: Restart bootstrap
[DBG ][m6LND]: app_parse_network_event() 8
[DBG ][m6LND]: ND/RPL scan new network
[DBG ][m6LND]: Restart bootstrap
[DBG ][m6LND]: app_parse_network_event() 8

gdillen, 2016-05-04:
I'm completely stuck. Anyone any idea what still can be wrong?

Many thanks.

gdillen, 2016-05-04:
Hello,

I'm one step closer: I have put the client and the gateway in the same room (closer to each other).
Now I get the following messages on the client. After the error 11 the client stops.

[INFO][m6LND]: 6LoWPAN ND bootstrap ready
[DBG ][m6LND]: ND Access Point: fd9f:b3fa:a2dd::ff:fe00:b448
[DBG ][m6LND]: ND Prefix 64: fd:9f:b3:fa:a2:dd:00:00
[DBG ][m6LND]: GP IPv6: fd9f:b3fa:a2dd:0:fec2:3d00:4:a5f8
[DBG ][m6LND]: MAC 16-bit: ff:ff
[DBG ][m6LND]: PAN ID: 06:91
[DBG ][m6LND]: MAC 64-bit: fc:c2:3d:00:00:04:a5:f8
[DBG ][m6LND]: IID (Based on MAC 64-bit address): fe:c2:3d:00:00:04:a5:f8
[DBG ][m6LND]: Channel: 12
                          mesh_network_handler() 0
[DBG ][ns_wrap]: ns_wrapper_socket_open(3)
                                          waiting 5s before sending registration...
send_registration()
[DBG ][ns_wrap]: SOCKET_TX_DONE, 163 bytes sent
[DBG ][ns_wrap]: SOCKET_TX_DONE, 163 bytes sent
[DBG ][ns_wrap]: SOCKET_TX_DONE, 163 bytes sent
                                               error 11

teetak01, 2016-05-05:
Hi @gdillen, can you check if you have valid security.h? From your logs it would seem that Client failed after it tried to initiate hand-shake with Connector and failed after few re-tries. You could cross-check also with our https://github.com/search?utf8=%E2%9C%93&q=mbed-client-quickstart which uses ethernet just to verify that you can get registration through to connector.

Compiling with "yt build -d" should give more traces out for debugging. Also make sure you have all the latest fixes (yotta update),

mikaleppanen, 2016-05-11:
Hi @gdillen,
Do you still have problems with the test setup or is it working? Is it ok to close this issue?
BR,
Mika

SeppoTakalo, 2016-05-20:
Closing this as the log file shows it succesfully joined the 6LoWPAN network.
Please re-open if the problem still exist.

closed 2016-05-20 by SeppoTakalo