ARMmbed/mbed-client-example-6lowpan #75: Unable to connect to 6lowpan network (app_parse_network_event 8)

shahab5, 2016-05-12:
Hi,

I have ported the code to a different PHY chip, this module is configured as 6lowpan client.
FireFly gateway router is used with Static binary (for testing purpose), goal is to reach the socket tx send state as mentioned in the example.

Few packet exchanges are seen on the air (attached sniffer capture), then nano-stack issues `Restart bootstrap` command.

With the same router setup, using FRDM board, I can see nano-stack issues many TX packets, unlike in my case. 

Why the device is not proceeding any further?
Kindly, let me know what's happening here. 

FYI, same application and nano-stack library are used on both the boards.

Thanks,
Shahab

**On FRDM board**  ( Tx packet with seq. num 0x07 can be seen )
[image]

**On my device** 
[image]

**Client Trace log**
                  Start mbed-client-example-6lowpan
[DBG ][m6La]: init()
[DBG ][m6La]: connect()
[DBG ][m6LND]: Link-layer security NOT enabled.
[DBG ][m6LND]: Channel: 12
[DBG ][m6LND]: Channel page: 0
[DBG ][m6LND]: Channel mask: 4096
[INFO][m6LND]: Start 6LoWPAN ND Bootstrap
[DBG ][m6LND]: app_parse_network_event() 8
[DBG ][m6LND]: ND/RPL scan new network
[DBG ][m6LND]: Restart bootstrap
[DBG ][m6LND]: app_parse_network_event() 8
[DBG ][m6LND]: ND/RPL scan new network
[DBG ][m6LND]: Restart bootstrap
[DBG ][m6LND]: app_parse_network_event() 8
[DBG ][m6LND]: ND/RPL scan new network
[DBG ][m6LND]: Restart bootstrap
[DBG ][m6LND]: app_parse_network_event() 8
[DBG ][m6LND]: ND/RPL scan new network
[DBG ][m6LND]: Restart bootstrap
[DBG ][m6LND]: app_parse_network_event() 8
[DBG ][m6LND]: ND/RPL scan new network
[DBG ][m6LND]: Restart bootstrap
[DBG ][m6LND]: app_parse_network_event() 8

shahab5, 2016-05-14:
Well, after observing the TX_STATUS at each packet, I made a fix.

For a TX_PACKET with ACK_REQ set, on my module only `PHY_TX_DONE` was sent (after ACK was received) to `arm_net_phy_tx_done()`.

While on FRDM board, it sends `PHY_TX_SUCCESS` followed by `PHY_TX_DONE`.

With this fix, Tx packet is sent.
Now, I get `GP REG fail`.

**Client debug trace**
    Start mbed-client-example-6lowpan
[DBG ][m6La]: init()
[DBG ][m6La]: connect()
[DBG ][m6LND]: Link-layer security NOT enabled.
[DBG ][m6LND]: Channel: 12
[DBG ][m6LND]: Channel page: 0
[DBG ][m6LND]: Channel mask: 4096
[INFO][m6LND]: Start 6LoWPAN ND Bootstrap
[DBG ][m6LND]: app_parse_network_event() 4
[DBG ][m6LND]: ND Scan/ GP REG fail
[DBG ][m6LND]: Restart bootstrap
[DBG ][m6LND]: app_parse_network_event() 4
[DBG ][m6LND]: ND Scan/ GP REG fail
[DBG ][m6LND]: Restart bootstrap
[DBG ][m6LND]: app_parse_network_event() 4
[DBG ][m6LND]: ND Scan/ GP REG fail
[DBG ][m6LND]: Restart bootstrap
[DBG ][m6LND]: app_parse_network_event() 4
[DBG ][m6LND]: ND Scan/ GP REG fail
[DBG ][m6LND]: Restart bootstrap

**Sniffer Capture**
[image]
[image]
[image]

kjbracey-arm, 2016-05-16:
The capture in pcap format would be useful. 

The error from the stack indicates one of two possible errors - failure get an ICMPv6 Router Advertisement in response to ICMPv6 Router Solicitations, or a failure to get a response to its address registrations (Address Registration Option in an ICMPv6 Neighbour Solicitation).

If I manually decode your trace correctly, packet 62 is an RS, and packets 63+65 are an RA response. But the node doesn't seem to act on the RA, and your trace suggests it didn't think it received it. (And, oddly, it also doesn't seem to respond to the incoming NS packets 67,69,71 - it really should, assuming it hadn't given up and shut down already).

Check your driver's reception side. There's no evidence of it passing up any packets after number 58. Maybe there's an issue with back-to-back fragmented packets 63+65.

shahab5, 2016-05-16:
Thank you for the reply.

Sorry, the capture is in .psd format (TI sniffer packet format).
Tried converting from PSD to PCAP, then wireshark is not interpreting the packets correctly [(psd2pcap)](https://e2e.ti.com/support/wireless_connectivity/low_power_rf_tools/f/155/t/56980l).

The Rx packets are sent to stack, added a dword print at
`rf_start_cca` (tx pkt first dword)
`arm_net_phy_tx_done` (txs -- tx status)
`arm_net_phy_rx` (rx pkt first dword)
`rf_interface_control` (RF state and channel)

**Client Debug Trace** 
[DBG ][m6LND]: Restart bootstrap
RF
2
12
tx ff600803
txs 2
rx 91c98000
RF
1
0
RF
2
12
tx 9161c861
txs 2
txs 0
rx 91bc8841
tx 9162c861
txs 2
txs 0
rx 91bdcc61
tx 9163cc61
txs 2
txs 0
rx 91bec841
rx 91bf8841
tx 9164c841
txs 2
rx 91c09c71
rx 91c18c71
rx 91c29841
rx 91c38841
rx 91c48c61
rx 91c68c61
rx 91c78841
rx 91c88c61
rx 91c9c841
rx 91ca8c61
rx 91cb9841
rx 91cc8841
rx 91cd9841
rx 91ce8841
rx 91cf8841
rx 91d08c61
rx 91d1c841
rx 91d2c841
rx 91d3c841
rx 91d48841
rx 91d5c841
RF
1
0
[DBG ][m6LND]: app_parse_network_event() 4
[DBG ][m6LND]: ND Scan/ GP REG fail

**Sniffer capture in pcap & psd format**
[client_board.pcap.zip](https://github.com/ARMmbed/mbed-client-example-6lowpan/files/266371/client_board.pcap.zip)
[client_board.psd.zip](https://github.com/ARMmbed/mbed-client-example-6lowpan/files/266372/client_board.psd.zip)

shahab5, 2016-05-19:
Hi @kjbracey-arm

 Please let me know, if anything else can be done to debug this issue.
 As, updated in ealier comment, the nano-stack gives up (RF 1 0 -- PHY_SHUT_DOWN) only after receiving the packets.

Thanks,
Shahab

shahab5, 2016-05-24:
Hi,

Can someone please suggest a solution?

Thanks,
Shahab

SeppoTakalo, 2016-05-26:
Hi @shahab5 

I have two theories here, first is that maybe the stack hangs up in the TX state and is waiting for proper callback to `arm_net_phy_tx_done()`

Please see the: https://docs.mbed.com/docs/arm-ipv66lowpan-stack/en/latest/16_API_porting/index.html#device-driver-api

Specifically:
[image]

So first it is required to send `SUCCESS` and the `TX_DONE`  (However, now looking at your comments I see that you already fixed this).

**My second theory is:** Your driver is now too slow, because you are printing stuff in your handlers. Network speed is 256kb/s so it is much faster that the serial port 115 that you are outputting your debug messages. During these prints, the packet timeouts may pass and you are losing packets that you otherwise may receive while printing.

Please verify that you are not printing anything from interrupt handlers, (or basically anywhere from the driver). If you want to print out stuff, you need to have specific event handler and send events through scheduler so you do not block the RF driver operation.

shahab5, 2016-05-26:
@SeppoTakalo Thanks for the suggestions. 
Disabled prints, alas! No change.

Packet capture shows that after RS Transmitted, no further Tx occurs.

Thanks,
Shahab

SeppoTakalo, 2016-05-26:
Can you try to capture the network traffic in a Wireshark somehow? Or convert the capture traffic to `.pcap` format, so could understand the packets?

shahab5, 2016-05-26:
Hi @SeppoTakalo PFA sniffer capture in `.pcap` format. 

[client_board.pcap.zip](https://github.com/ARMmbed/mbed-client-example-6lowpan/files/284420/client_board.pcap.zip)

SeppoTakalo, 2016-05-26:
Btw. We have used these sniffers http://www.ubisys.de/en/engineering/products-wireshark-802.15.4.html

They support Wireshark and 802.15.4 2.4 GHz network

shahab5, 2016-05-26:
Thanks for this info. I have been using TI's CC2351 based dongle, will try to procure some of these

SeppoTakalo, 2016-05-26:
Thanks for the capture.

It seems that your device starts a proper handshake with a border router.

Sequence looks like this:

1. Send: Beacon request
2. RX: Beacon
3. reset radio
4. 3-way MLE handshake
  1. TX: Link request
  2. RX: Accept&Request
  3. TX: Accept
5. TX: Router Solicitation
6. RX: 2 packets, fragmented Router Advertisement. This is where problems starts to appear.

After the step 5, the device does not seem to send anything. It should send Router Solicitation again or Neighbour Solicitation with global addresses.

[image]

Can you verify if stack is trying to send after the step 5. It should try to send those mentioned packets.

shahab5, 2016-05-26:
Indeed, it should.
However, from the prints it is clear that it isn't initiating any TX after the RS (65 packet).
No packets were received by the PHY driver.

How can I know if there an ICMPv6 error occurred on de-fragmentation? Does the stack reports an error in this case?

shahab5, 2016-06-08:
Hi,

I am still stuck on this. 
Can someone suggest anything else?

Thanks,
Shahab

SeppoTakalo, 2016-06-09:
Hi @shahab5 
Do you have access to Nanostack source code?
If so, you might want to try to enable debugging traces and see what it is doing.
I believe most of the traces are enabled when you build with `yotta build -d`

shahab5, 2016-06-10:
Hi @SeppoTakalo 
Unfortunately, I don't have access to Nanostack source code.
Are there any debug binaries available for use?

Thanks,
Shahab

shahab5, 2016-06-10:
What is the trigger event for the next Tx packet (Neighbour Solicitation) to be sent by nanostack?
Besides the Rx packets, is there any other thing which should occur for the Tx ?

Thanks,
Shahab

stevew817, 2016-10-10:
Was this issue ever resolved? I'm seeing the same thing on my end when using the LOWPAN_HOST binary and our own RF PHY. Nanostack bails with a [DBG ][m6LND]: app_parse_network_event() 8.
Using LOWPAN_ROUTER works fine.

SeppoTakalo, 2016-10-10:
@stevew817 
This is probably a separate issue.
When using `LOWPAN_HOST` binary, you should set Mesh API to use also host mode instead of router (default).
See: https://github.com/ARMmbed/mbed-os/tree/master/features/nanostack/FEATURE_NANOSTACK/mbed-mesh-api#module-configuration

So in `mbed_app.json` you should set the key: `mbed-mesh-api.6lowpan-nd-device-type` to value `LOWPAN_HOST`

stevew817, 2016-10-10:
@SeppoTakalo Thanks for taking the time to answer! I just (about a minute ago) figured out the same thing by tracing function calls. Maybe this deserves a mention in the examples? Perhaps by adding `"mbed-mesh-api.6lowpan-nd-device-type" : "NET_6LOWPAN_ROUTER"` as a verbose option in the example's config file (where LOWPAN_ROUTER in the features also needs to be changed to LOWPAN_HOST) instead of silently letting it default to router?

SeppoTakalo, 2016-10-10:
@stevew817 Yes. You are right, this needs to be documented.
We are in process of integrating the mbed OS 5.2 stuff into master and once we release that, we update our examples also. They will mention this ROUTER/HOST switching.

teetak01, 2019-04-24:
Closing, repository is being archived.

closed 2019-04-24 by teetak01