ARMmbed/mbed-client-example-6lowpan #76: IPv6 for 6LoWPAN or Thread border router over IPv4 based ISP

shahab5, 2016-05-17:
Hi,

My ISP provides IPv4 only and so a local IPv6 network using a Dlink router is created.
The border router is connected to this network.
But the `Ethernet bootstrap Ready` message is not seen.

FYI, using the gateway6LoWPANDynamic.bin.

Please suggest a way to work with IPv4 based ISP.

Thanks,
Shahab

teetak01, 2016-05-17:
Hi @shahab5, you would have to use some kind of "IPv6 tunnel broker" service if you want to connect to Connector-service via IPv4 network.

shahab5, 2016-05-17:
@teetak01 Thanks will try and update

SeppoTakalo, 2016-05-17:
@shahab5 The border router image you use, do not continue to set up the network, unless it receives a ICMPv6 Router Advertisement from the IPv6 capable router.
Therefore the "Bootstrap complete" message is not seen.

I would suggest that you try to set up a IPv6 tunnel of some sort. Some have done it with separate Linux box providing the tunnel, some have a Wifi gateway with capability to do so. I'm not familiar of any other tunnel providers than https://tunnelbroker.net/ who provides free tunnel.

SeppoTakalo, 2016-05-17:
I'm closing this as it is basically not a issue with a mbed-client but more generally a network infrastructure problem.
This is a known problem for us, and tunneling the IPv4 over IPv6 is the only solution. Mesh networks based on 6LoWPAN do not work on IPv4 addresses, therefore the must for IPv6.
IPv4 address space would be exhausted very quickly for mesh type of networks.

Br.
Seppo

closed 2016-05-17 by SeppoTakalo

shahab5, 2016-05-20:
Thanks for all the answers.
Using tunnelbroker.net, it worked.

Thanks,
Shahab