ARMmbed/mbed-client-example-6lowpan #77: What roles of the Thread network are demonstrated by this demo?

shahab5, 2016-05-19:
HI,

I need to understand what roles are demonstrated by this example. Since there are only two modules used here, Border Router and Joiner Router.

**Question**
 Is the Thread Gateway (Border) Router acting as Leader and Commisioner as well? 

Kindly, help me understand.

Thanks,
Shahab

artokin, 2016-05-23:
Hi,

The first router enabled device started will begin as a Leader of the Thread network (End Device can't be a Leader).

In this example we have two devices and they are using static configuration to establish a Thread network. When static configuration is used then required network parameters are already known by the joining device and therefore commissioning process is not needed.

BR,
Arto

shahab5, 2016-05-26:
Thank you very much for the explanation.

Is there a plan to have a Thread demo with commissioning process?

Regards,
Shahab

mlnx, 2016-05-26:
Shahab,

> Is there a plan to have a Thread demo with commissioning process?

Yes, there will be a demo at some point in the future.
If you are an mbed Partner, please contact your Partner Lead for more information.

Best regards,
Marcelo

shahab5, 2016-05-26:
Thank you all.
Closing this issue.

Regards,
Shahab

closed 2016-05-26 by shahab5