ARMmbed/mbed-client-example-6lowpan #78: Unable to connect to thread network

chikkaram, 2016-05-19:
Hi,
I have ported the code to a different PHY chip, this module is configured as thread client.
My device is unable to connect to mbed connector, Can you please help me.
Please find the below debug trace.

Debug trace:
=====================
[DBG ][mClt]: M2MNsdlInterface::create_nsdl_resource_structure(object_name Test/0)
[DBG ][mClt]: M2MNsdlInterface::create_nsdl_resource_structure - res_name Test/0/D
[DBG ][mClt]: M2MNsdlInterface::create_nsdl_resource(name Test/0/D)
[DBG ][mClt]: M2MBase::set_under_observation - observed: 0
[DBG ][mClt]: M2MBase::set_under_observation - base_type: 1
[DBG ][mbed-client]: M2MReportHandler::M2MReportHandler()
[DBG ][mbed-client]: M2MReportHandler::set_under_observation(observed 0)
[DBG ][mbed-client]: M2MReportHandler::stop_timers()
[DBG ][mbed-client]: M2MReportHandler::stop_timers() - out
[DBG ][mbed-client]: M2MReportHandler::set_default_values
[DBG ][mClt]: M2MNsdlInterface::create_nsdl_resource - Creating in NSDL-C result 0
[DBG ][mClt]: M2MBase::set_under_observation - observed: 0
[DBG ][mClt]: M2MBase::set_under_observation - base_type: 1
[DBG ][mbed-client]: M2MReportHandler::set_under_observation(observed 0)
[DBG ][mbed-client]: M2MReportHandler::stop_timers()
[DBG ][mbed-client]: M2MReportHandler::stop_timers() - out
[DBG ][mbed-client]: M2MReportHandler::set_default_values
[DBG ][mClt]: M2MNsdlInterface::create_nsdl_resource_structure(object_name Test/0)
[DBG ][mClt]: M2MNsdlInterface::create_nsdl_resource_structure - res_name Test/0/S
[DBG ][mClt]: M2MNsdlInterface::create_nsdl_resource(name Test/0/S)
[DBG ][mClt]: M2MBase::set_under_observation - observed: 0
[DBG ][mClt]: M2MBase::set_under_observation - base_type: 1
[DBG ][mbed-client]: M2MReportHandler::M2MReportHandler()
[DBG ][mbed-client]: M2MReportHandler::set_under_observation(observed 0)
[DBG ][mbed-client]: M2MReportHandler::stop_timers()
[DBG ][mbed-client]: M2MReportHandler::stop_timers() - out
[DBG ][mbed-client]: M2MReportHandler::set_default_values
[DBG ][mClt]: M2MNsdlInterface::create_nsdl_resource - Creating in NSDL-C result 0
[DBG ][mClt]: M2MBase::set_under_observation - observed: 0
[DBG ][mClt]: M2MBase::set_under_observation - base_type: 1
[DBG ][mbed-client]: M2MReportHandler::set_under_observation(observed 0)
[DBG ][mbed-client]: M2MReportHandler::stop_timers()
[DBG ][mbed-client]: M2MReportHandler::stop_timers() - out
[DBG ][mbed-client]: M2MReportHandler::set_default_values
[DBG ][mClt]: M2MInterfaceImpl::state_register - create_nsdl_list_structure - success
[DBG ][mClt]: M2MInterfaceImpl::state_registe_registe
[DBG ][mClt]: M2MInterfaceImpl::state_register - server_address coap://2607:f0d0:2601:52::20:5684
[DBG ][mClt]: M2MInterfaceImpl::state_register - IP address 2607:f0d0:2601:52::20 , Port 5684
[ERR ][mClt]: M2MInterfaceImpl::state_register - set error as M2MInterface::NetworkError
[ERR ][mClt]: M2MInterfaceImpl::state_register - Error Occured 6
[DBG ][mClt]: M2MInterfaceImpl::internal_event : new state 0
[DBG ][mClt]: M2MInterfaceImpl::state_engine
[DBG ][mClt]: M2MNsdlInterface::stop_timers()
[DBG ][mClt]: M2MInterfaceImpl::state_idle
                                          error 6
Reconnecting to server
[DBG ][mClt]: M2MInterfaceImpl::register_object(M2MSecurity *security,const M2MObjectList &object_list) - OUT
[DBG ][m6Thread]: app_parse_network_event() 0

chikkaram, 2016-05-20:
Hi,
I'm attaching thread_client.zip, contains debug trace and packets capture.
Thanks
[thread_client.zip](https://github.com/ARMmbed/mbed-client-example-6lowpan/files/273916/thread_client.zip)

teetak01, 2016-05-20:
Hi @chikkaram, are you using IPv6 tunneling to connect your border router to internet or perhaps proper IPv6 network from your ISP?

Check this closed issue: https://github.com/ARMmbed/mbed-client-example-6lowpan/issues/76

chikkaram, 2016-05-20:
Hi,
I have already created thread network(which supports IPv6 tunneling ), with FRDM-K64F client device and FireFly gateway router, tested successfully with mbed connector.  
**I have ported the same client device code to a different PHY chip, this module is configured as thread client, but my module is unable to join the existing  network, goal is to reach the socket tx send state as mentioned in the example.**
Thanks

SeppoTakalo, 2016-05-26:
Hi @chikkaram do you have a RF sniffer to verify if you PHY device is able to send&receive packets from the network?

From the logs provided, I cannot detect if you have even joined the network at all.
Would you be able to produce full logs from the device, and preferably without those mbed-client debug messages because they may slow down the RF driver.

TuomoHautamaki, 2016-06-15:
Hi @chikkaram can you please reply to Seppo's comment so we can get this further. Thanks

SeppoTakalo, 2016-07-06:
Closing this as there has been no updates or feedback for a long time.
With the given information, we are not able to provide any support for this.

teetak01, 2019-04-24:
Closing, repository is being archived.

closed 2019-04-24 by teetak01