ARMmbed/mbed-client-example-6lowpan #79: arm_net_phy_tx_done returns -1

shahab5, 2016-05-25:
Hi,

Setup uses a FRDM board as mbed client in 6lowpan with a border router.
Although, it works and gets detected in mbed device connector. But I didn't understand the reason for the API to return -1.

FYI, `interface_id = 0` and `tx_handle = 1` (verified this at `rf_start_cca`).

Thanks,
Shahab

SeppoTakalo, 2016-05-25:
Is this related https://github.com/ARMmbed/mbed-client-example-6lowpan/issues/75

shahab5, 2016-05-26:
During the debugging of #75, I observed this API throws an error both on my platform and FRDM-K64F. However, the latter board works fine.

SeppoTakalo, 2016-05-26:
@shahab5 Thanks for reporting this.

I checked the code and it seems that the `arm_net_phy_tx_done()` returns `-1` always, unless given TX handle is incorrect when it would return `-2`

This is now taken into our backlog and will be fixed in future releases.
As a workaround, you can ignore the return value for now.

Br. Seppo.

SeppoTakalo, 2016-06-07:
I will close this from this example applications. This has been recorded into our Nanostack's backlog.

closed 2016-06-07 by SeppoTakalo