ARMmbed/mbed-client-example-6lowpan #80: Static 6lowpan and error 11

svastm, 2016-05-31:
I have some trouble using the static 6lowpan example with the `DeviceServer v2.2.0-606` on my computer. Is it supported ? I use the frdm-k64f board with firefly shield and gateway.

the server starting well :
```
Device Server Developer (starting..)
Version: 2.2.0-606
CoAP over UDP interface [listening on /0.0.0.0:5683]
HTTP interface [listening on /0.0.0.0:8080]
Admin-UI HTTPS interface [listening on /0.0.0.0:8081]
Started.
```

But the node don't receive the `SOCKET_DATA`.
```
Start mbed-client-example-6lowpan
[DBG ][m6La]: init()
[DBG ][m6La]: connect()
[DBG ][m6LND]: Link-layer security NOT enabled.
[DBG ][m6LND]: Channel: 12
[DBG ][m6LND]: Channel page: 0
[DBG ][m6LND]: Channel mask: 4096
[INFO][m6LND]: Start 6LoWPAN ND Bootstrap
[DBG ][m6LND]: app_parse_network_event() 0
[INFO][m6LND]: 6LoWPAN ND bootstrap ready
[DBG ][m6LND]: ND Access Point: fd00:ff1:ce0b:a5e0:0:ff:fe00:3ea5
[DBG ][m6LND]: ND Prefix 64: fd:00:0f:f1:ce:0b:a5:e0
[DBG ][m6LND]: GP IPv6: fd00:ff1:ce0b:a5e0:fec2:3d00:4:879e
[DBG ][m6LND]: MAC 16-bit: ff:ff
[DBG ][m6LND]: PAN ID: 06:91
[DBG ][m6LND]: MAC 64-bit: fc:c2:3d:00:00:04:87:9e
[DBG ][m6LND]: IID (Based on MAC 64-bit address): fe:c2:3d:00:00:04:87:9e
[DBG ][m6LND]: Channel: 12 mesh_network_handler() 0
[DBG ][ns_wrap]: ns_wrapper_socket_open(1)
waiting 5s before sending registration...
send_registration()
[DBG ][ns_wrap]: SOCKET_TX_DONE, 163 bytes sent
[DBG ][ns_wrap]: SOCKET_TX_DONE, 163 bytes sent
[DBG ][ns_wrap]: SOCKET_TX_DONE, 163 bytes sent
[DBG ][ns_wrap]: ns_wrapper_socket_free(1)
[DBG ][ns_wrap]: ns_wrapper_socket_open(2)
[DBG ][ns_wrap]: SOCKET_TX_DONE, 163 bytes sent
[DBG ][ns_wrap]: SOCKET_TX_DONE, 163 bytes sent
[DBG ][ns_wrap]: SOCKET_TX_DONE, 163 bytes sent
[DBG ][ns_wrap]: ns_wrapper_socket_free(2)
[DBG ][ns_wrap]: ns_wrapper_socket_open(3)
[DBG ][ns_wrap]: SOCKET_TX_DONE, 163 bytes sent
[DBG ][ns_wrap]: SOCKET_TX_DONE, 163 bytes sent
[DBG ][ns_wrap]: SOCKET_TX_DONE, 163 bytes sent
[DBG ][ns_wrap]: ns_wrapper_socket_free(3)
[DBG ][ns_wrap]: ns_wrapper_socket_open(4)
[DBG ][ns_wrap]: SOCKET_TX_DONE, 163 bytes sent
[DBG ][ns_wrap]: SOCKET_TX_DONE, 163 bytes sent
[DBG ][ns_wrap]: SOCKET_TX_DONE, 163 bytes sent
error 11
```
The error code `11` refer to `SecureConnectionFailed`. I have fill my `security.h` file with my API keys, found on the ARM device connector website. The error might be here but the strange thing is even if the server is shutdown, the error `11` is triggered.

gateway log:
```
P.Init
[DBG ][pana]: Init
[DBG ][sck ]: Socket id 0 allocated
Starting NanoRouter in static mode, extension=1
[DBG ][core]: NS Root task Init
[DBG ][sck ]: Socket Tasklet Generated
[DBG ][mle ]: MLE Tasklet Generated
[DBG ][pana]: Pana Tasklet Init
[DBG ][board]: Setting up TX power to be 0x0F+Amp = +4.1 dBm
[INFO][EP  ]: endpoint_tasklet(): init
[DBG ][sck ]: Socket id 1 allocated
[... 161 lines cut ...]
Router IP Addr: fd00:ff1:ce0b:a5e0:fec2:3d00:4:879e
LL Addr: (REACHABLE 1419) 06:91:fc:c2:3d:00:00:04:87:9e
Router IP Addr: fe80::fec2:3d00:4:879e
LL Addr: (REACHABLE 1219) 06:91:fc:c2:3d:00:00:04:87:9e
...
```

ipconfig:
```
Connection-specific DNS Suffix  . :
IPv6 Address. . . . . . . . . . . : fd00:ff1:ce0b:a5e0::1
IPv6 Address. . . . . . . . . . . : fd00:ff1:ce0b:a5e0:d1b:b2fa:d5aa:e61f
Temporary IPv6 Address. . . . . . : fd00:ff1:ce0b:a5e0:ddbb:5fc2:d54c:1211
Link-local IPv6 Address . . . . . : fe80::d1b:b2fa:d5aa:e61f%11
Autoconfiguration IPv4 Address. . : 169.254.230.31
Subnet Mask . . . . . . . . . . . : 255.255.0.0
Default Gateway . . . . . . . . . :
```

in the config.json file:
```javascript
"device_connector_uri": "\"coap:\/\/fd00:ff1:ce0b:a5e0::1:5683\"",
```

I followed the update of the `README` by @hasnainvirk.

SeppoTakalo, 2016-05-31:
Hi @svastm 
You cannot use security certificates generated by connector service for your own standalone mDS. Those are valid security certificates that connector uses to validate your device. Your standalone mDS have no knowledge of these certificates, therefore it should be rejecting them.

When testing with local server instance, all security features should be disabled.

svastm, 2016-05-31:
Ok I have missed some change in the updated branch. Now the security layer is disable by changing
```c
security->set_resource_value(M2MSecurity::SecurityMode, M2MSecurity::Certificate);
```
 to 
```c
security->set_resource_value(M2MSecurity::SecurityMode, M2MSecurity::NoSecurity);
```
 in `mbedclient.c`. I assume that the default config of coAP for the mDS is working without any security (at least on UDP port 5683).

This port 5683 looks open in windows.
```
>netstat -an | find "5683"
  UDP    0.0.0.0:5683           *:*
  UDP    [::]:5683              *:*
```
I'm still stuck at the same point, the error code changed to `6` aka `NetworkError`.

hasnainvirk, 2016-06-01:
@svastm 5683 is a secure port. Please use 5684 if you are not using secure communication.

kjbracey-arm, 2016-06-01:
Wrong way round - 5684 is secure.

Note that current code ignores the coap/coaps in the URL, and determines security entirely from the separate M2MSecurity API. The URL parsing is also not totally solid, but I suggest using a valid syntax that it does accept: coap://[fd00:ff1:ce0b:a5e0::1]:5683

IPv6 literal URLs should have the [], but older versions of the code didn't support this, which is why some examples omit it. The port shouldn't need to be specified, but current code requires it.

svastm, 2016-06-01:
the [] trick did not solve the problem.
Because it's now marked as a network error, I opened wireshark and saw only ICMPv6 neighbor solicitation and advertisement but no coAP packet. So the problem don't come from the mDS config. I have tried to change the `sal-stack-nanostack` from `4.1.0` to `4.0.3` because the last update of GW_Binary was for this version but i still have the same result.

hasnainvirk, 2016-06-01:
can you please show me the output of "route -6 -n" command ?

hasnainvirk, 2016-06-01:
Also attach wirepas snapshots if possible.

SeppoTakalo, 2016-06-01:
After those latest changes, are you able to see `6Lowpan bootstrap ready, IP=...` message in clients log?

svastm, 2016-06-01:
Yes the message `6Lowpan bootstrap ready, IP=...` is still present in the gateway logs.
```
> route PRINT -6
===========================================================================
Interface List
 11...08 2e 5f 1b 04 f5 ......Intel(R) 82579LM Gigabit Network Connection
  1...........................Software Loopback Interface 1
 13...00 00 00 00 00 00 00 e0 Microsoft ISATAP Adapter #2
===========================================================================

IPv6 Route Table
===========================================================================
Active Routes:
 If Metric Network Destination      Gateway
 11    276 fd00:ff1:ce0b:a5e0::/64  On-link
 11    276 fd00:ff1:ce0b:a5e0::1/128
                                    On-link
 11    276 fd00:ff1:ce0b:a5e0:d1b:b2fa:d5aa:e61f/128
                                    On-link
 11    276 fd00:ff1:ce0b:a5e0:2c1f:9d5e:a05a:51ec/128
                                    On-link
 11    276 fe80::/64                On-link
 11    276 fe80::d1b:b2fa:d5aa:e61f/128
                                    On-link
 11    276 ff00::/8                 On-link
===========================================================================
Persistent Routes:
  None

```

svastm, 2016-06-01:
Here a pcap of the icmpv6 packets. It begin after a reset of the gateway [6lowpan.zip](https://github.com/ARMmbed/mbed-client-example-6lowpan/files/293456/6lowpan.zip)

SeppoTakalo, 2016-06-01:
The `6Lowpan bootstrap ready, IP=...` should be present the client logs.

svastm, 2016-06-01:
In the client i have the following one:

```
...
[INFO][m6LND]: 6LoWPAN ND bootstrap ready
[DBG ][m6LND]: ND Access Point: fd00:ff1:ce0b:a5e0:0:ff:fe00:7a94
[DBG ][m6LND]: ND Prefix 64: fd:00:0f:f1:ce:0b:a5:e0
[DBG ][m6LND]: GP IPv6: fd00:ff1:ce0b:a5e0:fec2:3d00:4:879e
[DBG ][m6LND]: MAC 16-bit: ff:ff
[DBG ][m6LND]: PAN ID: 06:91
[DBG ][m6LND]: MAC 64-bit: fc:c2:3d:00:00:04:87:9e
[DBG ][m6LND]: IID (Based on MAC 64-bit address): fe:c2:3d:00:00:04:87:9e
[DBG ][m6LND]: Channel: 12
...
```

SeppoTakalo, 2016-06-07:
Hi @svastm 
Any updates on this one?
Seems that both your GW and client have succesfully established the network.

Can you try to ping the IPv6 addresses shown in log files:
`[DBG ][m6LND]: ND Access Point: fd00:ff1:ce0b:a5e0:0:ff:fe00:7a94` shows you the IPv6 address of the GW.
Try to ping that.

If it works, try pinging the 6LowPAN node address shown in line `[DBG ][m6LND]: GP IPv6: fd00:ff1:ce0b:a5e0:fec2:3d00:4:879e`

Br.

svastm, 2016-06-08:
Hello,
Does the GW implement ping ?
When i ping the gateway:
```
>ping -6 [fd00:ff1:ce0b:a5e0:0:ff:fe00:d708]

Pinging fd00:ff1:ce0b:a5e0:0:ff:fe00:d708 with 32 bytes of data:
General failure.
General failure.
General failure.
General failure.

Ping statistics for fd00:ff1:ce0b:a5e0:0:ff:fe00:d708:
    Packets: Sent = 4, Received = 0, Lost = 4 (100% loss),
```
But if I change the adress for a non existing one. I have a `Destination host unreachable` error instead of `General failure`. Same result for the ping of the client.

Should i see something in the GW when the client send bytes ?

SeppoTakalo, 2016-06-08:
Yes, the GW and Node should both answer the ICMP Echo request (ping).

I'm not familiar of Windows, so I'm unsure what is causing the `General failure` but I have seen lots of these failures due the corporate firewall policy of some sort.

Can you do the tests on a Linux machine, or a machine with no firewalls installed/enabled? (on Windows, if restrictive policies are in place, you cannot disable firewall, even when the UI shows that it is disable).

svastm, 2016-06-10:
Without firewall on an other machine, it works. So it was a firewall problem. Thank you very much for the support. I think you can mark this issue as resolved.

SeppoTakalo, 2016-06-10:
@svastm Nice to hear. Happy to help you.

closed 2016-06-10 by SeppoTakalo