ARMmbed/mbed-client-example-6lowpan #81: Cannot access http://connect.arm.com

betzw, 2016-06-01:
On the [device_server readme page](https://github.com/ARMmbed/mbed-client-example-6lowpan/tree/device_server#downloading-mbed-ds--arm-reference-app) you are referring to two software packages (mbed Device Server & Referennce / Home App) to download in order to setup the server side of the demo. Unfortunately the site http://connect.arm.com is not accessible from within my company (most likely due to our firewall) ending up with the following error:
```
An error occurred during a connection to connect.arm.com. 
SSL peer was unable to negotiate an acceptable set of security parameters. 
Error code: SSL_ERROR_HANDSHAKE_FAILURE_ALERT
```
Any idea, how I can get around this problem?

SeppoTakalo, 2016-06-01:
Hi @betzw 

I have no idea of why connect.arm.com is generating SSL errors.
The link uses `http://` address but can you try with `https://`

There might be some HTTP redirection.

However, as this is not directly bug in this specific software package, I'm closing this bug here.

Please report back, if you still have errors. I'll then forward the information to IT team.

closed 2016-06-01 by SeppoTakalo

betzw, 2016-06-01:
Pls. forward this to your IT team!

Using `http` or `https` doesn't change anything as both end up in a secure connection setup which fails as described above!
An additional information you can forward to them is, that we are using a Zscaler based firewall, which "messes" around with certificates ...
I am copying also our contact in ARM which might help us accelerate finding a solution for this issue!

cc @screamerbg

betzw, 2016-06-06:
cc @nikapov-ST