ARMmbed/mbed-client-example-6lowpan #82: Mbed 6LoWPAN gateway problem

LTEKuser01, 2016-07-14:
Hi all,

Just to have a whole picture:
I have build 6LoWPAN Thread network with 5 nodes.
I created additional resource (temperature) in the client code.
After some time (period cca. 1 day) system stops.
Serial communication with gateway stops. You can see this in log files: serial_GW_NOK_*.log
As the problem is probably in the gateway, trying to reach devices directly trough https://connector.mbed.com/, does not help (0 devices discovered).

You can filter working time of the system with the help of demo2.html
I also attached log file of mbed_connector_api.py: connector_api_err_13_07.log
If you will search the file by timestamp, format of the timestamp in the file is like 2016-07-06 10:09:49.

I'm using  gatewayThreadDynamic.bin.

What could be the cause?

Thanks,
Primoz Zajec
[log_files.zip](https://github.com/ARMmbed/mbed-client-example-6lowpan/files/363343/log_files.zip)

artokin, 2016-08-15:
Hi,
Sorry for long delay and thanks for reporting this issue. 
It looks like there is a memory leak in the border router. Logfile "serial_GW_NOK_12_07" contains few "_out of memory_" and "_No mem_" lines before the gateway stops working.
Best Regards,
Arto

TuomoHautamaki, 2016-10-06:
Status check done and planning done howto proceed/solve this issue. So far we haven't had enough bandwidth to  build similar testing setup for this sample application.

artokin, 2016-10-24:
Hi,

I was not able to reproduce the problem using the new gateway binaries available in https://github.com/ARMmbed/mbed-os-example-client/tree/master/GW_Binary. After 1 day of usage no memory leak was detected and memory consumption was steadily between 21-22kBytes while peak memory consumption was 26kBytes. I didn't try with old binaries that were originally used in this issue.

@LTEKuser01 , would you be able to try again with the new gateway binaries? If the problem still persists then we could arrange a special gateway binary that has more traces turned on to pinpoint the leak location.

Best regards,
Arto

artokin, 2016-11-08:
Hi,

Closing this issue as we can't reproduce the error with the new gateway binaries available in https://github.com/ARMmbed/mbed-os-example-client/tree/master/GW_Binary. 

This repository is superseded in mbed OS 5.2. Please file a new issue to repository https://github.com/ARMmbed/mbed-os-example-client if issues are found from border router binaries.

Best regards,
Arto

closed 2016-11-08 by artokin