ARMmbed/mbed-client-example-6lowpan #85: can not see connected devices under the website  https://connector.mbed.com/

haishengliu, 2016-08-16:
Hi,

By following this tutorial, I'm not able to see connected devices on https://connector.mbed.com/
Do we need the ISP supports fully IPV6? We do proceed a number of tests with different ISPs with no success. Could you please give us more detail to work around? 
Thank you very much for your help

Haisheng

- dowload the gatewayThreadDynamic.bin file on the 6loWPAN gateway router.
 - generate a security.h file
- build and download a binary file for the client

Here are the debug messages on 6loWPAN gateway router:

P.Init Starting NanoRouter in autonomous mode, extension=2 
[DBG ][core]: NS Root task Init 
[DBG ][sck ]: Socket Tasklet Generated 
[DBG ][mle ]: MLE Tasklet Generated 
[DBG ][board]: Setting up TX power to be 0x0F+Amp = +4.1 dBm 
[INFO][EP ]: endpoint_tasklet(): init 
[DBG ][sck ]: Socket id 0 allocated 
[DBG ][sck ]: Socket id 1 allocated 
[DBG ][ip6s]: IPV6 interface Base Ready [DBG ][rout]: Added route: 
[DBG ][rout]: fe80::/64 if:1 src:'Static' id:0 lifetime:infinite 
[DBG ][rout]: On-link (met 128) Ipv6 bootsrap started 
[INFO][addr]: Tentative Address added to IF 1: fe80::fec2:3dff:fe05:39f1 
[DBG ][addr]: LL64 Register OK! [INFO][addr]: DAD passed on IF 1: fe80::fec2:3dff:fe05:39f1 
[DBG ][ip6s]: Interface ID: 1, ipv6: fe80::fec2:3dff:fe05:39f1 
[DBG ][ip6s]: DAD OK 
[DBG ][ip6s]: Push RS to IPV6 
[DBG ][ip6s]: Push RS to IPV6 
[DBG ][ip6s]: Push RS to IPV6 
[DBG ][ip6s]: Push RS to IPV6 

Debug messages on 6loWPAN client:

Start mbed-client-example-6lowpan 
[DBG ][m6La]: init() 
[DBG ][m6La]: connect() 
[DBG ][m6Thread]: PANID defa 
[DBG ][m6Thread]: channel: 12 
[DBG ][m6Thread]: channel page: 0 
[DBG ][m6Thread]: channel mask: 134215680 
[DBG ][m6Thread]: Mesh prefix: fd:00:0d:b8:00:00:00:00 
[DBG ][m6Thread]: Master key: 00:11:22:33:44:55:66:77:88:99:aa:bb:cc:dd:ee:ff 
[DBG ][m6Thread]: PSKc: 00:11:22:33:44:55:66:77:88:99:aa:bb:cc:dd:ee:ff 
[DBG ][m6Thread]: PSKd: Secret password 
[INFO][m6Thread]: Start Thread bootstrap (SED mode)

artokin, 2016-08-19:
Hi,

I think your setup is correct but the gateway is not able to connect to the connector service. To me this looks like an ISP issue as gateway router is missing traces that indicates bootstrap completion. Are you able to ping api.connector.mbed.com from a laptop that is connected to your ISP?

btw, would you please try our latest (mbed OS 5) example application at: https://github.com/ARMmbed/mbed-os-example-client? It contains a lot of improvements.

Best regards,
Arto

haishengliu, 2016-08-19:
Hi, 
Thank you very much for your help.
Yes, I can ping  api.connector.mbed.com with a laptop (IPV4 & 6) connected to the same network. It works just fine. We don't get idea how to see the switch connected. We are wondering if it is necessary to have a local IPV6 network.   I am going to try another application as you indicated. 

Thanks again

Best regards

Haisheng

closed 2016-10-21 by haishengliu